open62541/open62541 #4344: v1.2.1 - UA Expert cannot connect - Bad Timeout error

jackybek, 2021-04-16:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v.1.2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Raspian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jpfr, 2021-04-16:
Please give enough detail to reproduce and detailed log output.

encryption yes/no?

jackybek, 2021-04-16:
Julius,
Yes.  There is encryption enabled in my settings using openSSL.

I set the UA_LOGLEVEL = 100, and uncomment the following statement :
**UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(uaServer1), 7681, 0, 0, NULL, NULL);**

I then try to use UAExpert to access the my OPCServer.  The error shown in UA_Expert is : "Could not connect to Server.  Bad Timeout"
There is a trace recorded:
[2021-04-17 02:17:51.311 (UTC+0800)] trace/network      Connection 12 | New TCP connection on server socket 5
[2021-04-17 02:17:51.311 (UTC+0800)] info/network       Connection 12 | New connection over TCP from 192.168.1.119
[2021-04-17 02:17:51.311 (UTC+0800)] trace/network      Connection 11 | Activity on the socket
[2021-04-17 02:17:51.312 (UTC+0800)] trace/network      Connection 11 | Received a packet.
**[2021-04-17 02:17:51.312 (UTC+0800)] trace/channel      Connection 11 | SecureChannel 0 | Process a HEL message**

However, when i use a Web socket client to access UA Server, it can connect.
17/4/2021 2:16:33 am  Try to connect ws://192.168.1.109:7681
17/4/2021 2:16:33 am  Connect success.

To confirm that this statement in BOLD is the cause, i excluded it and recompile, then try to access UA_Server using UAExpert.  No issue.  I also tried WebSocket Client.  This time websocket connection failed.  This is expected

jackybek, 2021-04-17:
I made some changes to the parameters as follows:
                        UA_UInt32 sendBufferSize = 1024 * 1024; // 1 Mbyte
                        UA_UInt32 recvBufferSize = 1024 * 1024; // 1 Mbyte
                        UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(uaServer1), 7681, sendBufferSize, recvBufferSize, &certificate, &privateKey);

where certificate and privateKey is the same instance when I use it to instantiate the UA_Server.  This is the same outcome.

[2021-04-17 09:59:03.604 (UTC+0800)] trace/network      Connection 12 | Activity on the socket
[2021-04-17 09:59:03.604 (UTC+0800)] trace/network      Connection 12 | Received a packet.
**[2021-04-17 09:59:03.604 (UTC+0800)] trace/channel      Connection 12 | SecureChannel 0 | Process a HEL message**
[2021-04-17 09:59:03.605 (UTC+0800)] trace/network      Connection 11 | Activity on the socket
[2021-04-17 09:59:03.605 (UTC+0800)] info/network       Connection 11 | Closed

I am still not able to use UAExpert to connect my OPCServer.

I checked the definition in open62541.h.  Even if i did not set the values for the last 4 parameters, it will use default values.   In this case, my earlier command was : **UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(uaServer1), 7681, 0, 0, NULL, NULL);**

jpfr, 2021-04-19:
Changing the title to mention websockets

jackybek, 2021-12-29:
Issue is resolved in v1.3

jackybek, 2022-02-09:
When trying to use UaExpert to connect to a OPCUA server with the following:

     retval = UA_ServerConfig_addNetworkLayerWS(UA_Server_getConfig(uaServer1), 7681, sendBufferSize, recvBufferSize, &certificate, &privateKey);

I get this during runtime:

     [2022-02-09 21:50:35.530 (UTC+0800)] info/network       TCP network layer listening on opc.tcp://OPCServerUAT-115:4840/
     [2022-02-09 21:50:35.530 (UTC+0800)] info/network       Websocket network layer listening on ws://OPCServerUAT-115:7681/
     [2022/02/09 21:50:35:5308] N: LWS: 4.2.99-v4.2.0-178-g656a9c82, NET CLI SRV H1 H2 WS ConMon IPv6-absent
     [2022/02/09 21:50:35:5311] N:  ++ [wsi|0|pipe] (1)
     [2022/02/09 21:50:35:5312] N:  ++ [vh|0|netlink] (1)
     [2022/02/09 21:50:35:5314] N:  ++ [vh|1|ws://OPCServerUAT-115:7681/||7681] (2)
     [2022/02/09 21:50:35:5315] N: [null wsi]: lws_socket_bind: source ads 0.0.0.0
     [2022/02/09 21:50:35:5316] N:  ++ [wsi|1|listen|ws://OPCServerUAT-115:7681/||7681] (2)

At this juncture, if i use a websocket client to connect ws://192.168.1.115:7681, it is successfully connected. However, UaExpert get "Could not connect to server:BadTimeout.

if i comment out UA_ServerConfig_addNetworkLayerWS(), UaExpert can now connect to my OPCUA Server.

andreasebner, 2025-07-30:
Networking has changed drastically ever since the event loop landed, so I'll close this now. Feel free to reopen if this issue persists.