open62541/open62541 #4348: How to publish Dataset MetaData

MaoniinoaM, 2021-04-18:
Hi, Open62541 teams,
I'm trying to use pubsub to push something to Mindsphere with open62541.
One of the communition steps is to upload the dataset metadata to the cloud.
Actually I have set the PUBLISHER_METADATAUPDATETIME to 1 and PUBLISHER_METADATAQUEUENAME to right topic according to the example of example_pubsub_publish.c.
But it seems not to send the metadata out with wireshark monitoring.

I'm not sure if I'm right that PUBLISHER_METADATAUPDATETIME set to 1 will trigger open62541 send metadata out automatically.

If not, does open62541 provide any examples to implement this feature? I didn't figure out related example in the repo.

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

AlexexanderSavinov, 2021-06-03:
I also have the same question regarding the metaData publishing over MQTT. The OPC Specification states that the MetaData can be send over the metadata queue. I defined this queue as in the example. I Also see the MetaData being created when i usee UaExpert. What i still dont understand is how to trigger the publishing? 

Is it not implemented or am i missing something? At least there is no example available that is showing this.

in tutorial_pubsub_mqtt_publish there is even the following comment:

/*TODO: Publish Metadata to metaDataQueueName */

shaohuazhang1, 2022-09-27:
What does "set the PUBLISHER_METADATAUPDATETIME to 1"mean?

sudojh, 2024-07-18:
Encountering the same problem. Is there any solution by now?