open62541/open62541 #4353: Amalgamated files failled

AndrUser65, 2021-04-20:
## Description

I use current git-branch and Windows. In Cmake i choose UA_ENABLE_AMALGAMATION and compiled this project. New 
open62541.c i use and include in my project. But i get following errors.:
1) first error was C4996	`'inet_addr': Use inet_pton() or InetPton() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warning`  - i included _WINSOCK_DEPRECATED_NO_WARNINGS 

2)c4702 local variable is not initialized  `const unsigned char *label;`  i changed it to `const unsigned char *label=NULL;`

3)after that i get this 
## Background Information / Reproduction Steps

[image]

Commandline options:
-DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DUA_ENABLE_DISCOVERY:BOOL="1" -DUA_ENABLE_PUBSUB:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="1" -DUA_BUILD_TOOLS:BOOL="1" -DUA_ENABLE_JSON_ENCODING:BOOL="1" -DUA_ENABLE_DISCOVERY_MULTICAST:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" -DUA_ENABLE_PUBSUB_MONITORING:BOOL="1" -DUA_ENABLE_HISTORIZING:BOOL="1" -DUA_ENABLE_ENCRYPTION:BOOL="0" -DUA_ENABLE_PUBSUB_MQTT:BOOL="1" 

Cache file:
UA_NAMESPACE_ZERO:STRING=FULL
UA_ENABLE_AMALGAMATION:BOOL=1
UA_ENABLE_DISCOVERY:BOOL=1
UA_ENABLE_PUBSUB:BOOL=1
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=1
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=1
UA_BUILD_TOOLS:BOOL=1
UA_ENABLE_JSON_ENCODING:BOOL=1
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=1
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=1
UA_ENABLE_PUBSUB_MONITORING:BOOL=1
UA_ENABLE_HISTORIZING:BOOL=1
UA_ENABLE_ENCRYPTION:BOOL=0
UA_ENABLE_PUBSUB_MQTT:BOOL=1

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): current branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

EvgKlin, 2021-04-21:
i get the same problem. amalgamation do not work

happybruce, 2023-02-22:
Which version you are using?

closed 2026-10-20 by o6-nbeier