open62541/open62541 #4355: UA_ServerConfig_setDefaultWithSecurityPolicies problem with OPEN62541_FEERTOS_USE_OWN_MEM

embbo, 2021-04-20:
Hi all, I am using STM  and own Malloc funtions , My code funtions okay but when i am trying to use 
 #define OPEN62541_FEERTOS_USE_OWN_MEM

 than this funtion:

UA_ServerConfig_setDefaultWithSecurityPolicies(uaServerConfig,
			 portNumber,
			 &certificate,     &privateKey,
			 trustList,        trustListSize,
			 issuerList,       issuerListSize,
			 revocationList,   revocationListSize);

 is not working ? what can be problem ? 

"Could not connect to server: BadCommicationError"

--> without that funtion (UA_ServerConfig_setDefaultWithSecurityPolicies) I can connect with sever
           

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Arthurmgwork, 2021-05-12:
Heey, I have the same problem. But I use the esp32 with esp-idf 4.2 framework. I can't stabilish a Basic256Sha256 Encrypt a Minimal Server. I generate the certificate and key, but don't working. Anyone help?

Code of Task:

static void opcua_task(void *arg)
{
    //BufferSize's got to be decreased due to latest refactorings in open62541 v1.2rc.
    UA_Int32  sendBufferSize = 16000;
    UA_Int32  recvBufferSize = sendBufferSize ;
    UA_UInt16 portNumber     = 4840;

    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));

    ESP_LOGI(TAG, "Fire up OPC UA Server.");
    UA_Server *server = UA_Server_new(); ///////////////////////////////////////////

    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setMinimalCustomBuffer(config, portNumber, 0, sendBufferSize, recvBufferSize);

    char* certificate = CERTIFICADO;
    char* privateKey  = PRIVATE_KEY;

    size_t length_certificate = (sizeof(CERTIFICADO)/sizeof(char));
    size_t length_private_key = (sizeof(PRIVATE_KEY)/sizeof(char));

    ESP_LOGW("Encrypt", "Length_Certificate(%zu) Length_Private_Key(%zu)",length_certificate,length_private_key);

    UA_String certificado_STRING;
    UA_String key_STRING;

    certificado_STRING.length = length_certificate;
    certificado_STRING.data = pvPortMalloc(sizeof(UA_Byte)*length_certificate);
    for(int i=0;i<length_certificate;i++)
    {
        certificado_STRING.data[i] = certificate[i];
    }

    key_STRING.length = length_private_key;
    key_STRING.data = pvPortMalloc(sizeof(UA_Byte)*length_private_key);
    for(int i=0;i<length_certificate;i++)
    {
        key_STRING.data[i] = privateKey[i];
    }

    const UA_ByteString certificado_BS = certificado_STRING;
    const UA_ByteString key_BS = key_STRING;

    UA_ServerConfig_setDefaultWithSecurityPolicies(s_config, 
                                                       4840, 
                                                       &certificado_BS,
                                                       &key_BS,
                                                       0,//trustlist,
                                                       0,//trustlistSize,
                                                       0,//issuerList,
                                                       0,//issuerListSize,
                                                       0,//revocationList,
                                                       0);//revocationListSize);

    const char *appUri = "esp32";
    UA_String hostName = UA_STRING("esp32");

   UA_ServerConfig_setUriName(config, appUri, "esp32");
    UA_ServerConfig_setCustomHostname(config, hostName);

   UA_StatusCode retval = UA_Server_run_startup(server);

   Init_OPC_UA_Queue();
   if (retval == UA_STATUSCODE_GOOD)
   {
        while (running)
        {
            UA_Server_run_iterate(server, true);
            ESP_ERROR_CHECK(esp_task_wdt_reset());
            taskYIELD();
            
        }
        UA_Server_run_shutdown(server);
        
   }
    
   UA_Server_delete(server);
   ESP_ERROR_CHECK(esp_task_wdt_delete(NULL));

}//End Task

EvgKlin, 2021-06-25:
@Arthurmgwork @embbo  Do you find Solution?

embbo, 2021-07-06:
@EvgKlin 1.  browsing works fine without Encryption 
                 2.  what i found ,problem is here :
                  void *pvPortCalloc(size_t count, size_t size)
                void *pvPortRealloc(void *mem, size_t newsize)
if you can look these two function in heap ..

closed 2026-10-20 by o6-nbeier