open62541/open62541 #4359: Publish encoding APIs

jonasgreen88, 2021-04-21:
## Description

I would like to suggest to expose the APIs of the encoding implementations (currently UA binary and JSON) of the library. These encoding and decoding functions can be used by other purposes, for instance to use the encoding to read/write data of a file or exchange data (transport != OPC UA, e.g. MQTT) using a standardized encoding.

Does anyone see a problem in exposing these APIs? If not, I can make a PR.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): N/A
 - [x] Operating system: N/A
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-23:
Sounds good. But the public API needs to be cleaned up compared to the internal one.

Can you make a proposal?

jpfr, 2021-04-23:
Then we could also convert the shell tool

/tools/ua2json

to use only the public api.

jonasgreen88, 2021-04-23:
Sure, we can probably do a try sometime soon. Can we start with the Binary API and take care of the JSON API in a second PR?
When looking in the JSON API I can see that file can probably be separated into a private and a public part and it needs documentation. Can you give more specific examples what you think need to be updated in the Binary API? I guess also some of the generated file need to be mapped as public headers.

jpfr, 2021-08-16:
The binary encoding is now public API: #4597
JSON comes via the PR of @matkonnerth #4570

closed 2022-04-11 by jonasgreen88