open62541/open62541 #4369: Connecting to server_ctt with OpenSSL (Sign&Encrypt, Basic256sha256) from Windows 10

ixfg9922, 2021-04-26:
## Description

I'm trying to use an open62541 client running on Windows 10 in order to connect via OpenSSL (Sign&Encypt / basic256sha256) to an instance of the sample server for the open62541 Certified SDK. I've already been able to connect using what is virtually the same code from both Linux and Mac OS, but never from Windows.

I have an idea as to what may be causing the problem, but I'm not sure.

## Background Information / Reproduction Steps

When attempting to connect, the software that I'm writing calls UA_Client_connectUsername() and produces the following output:

```
Mon Apr 26 12:26:06 2021 [3132/47] * info/client	 Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/48] * warn/client	 The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
Mon Apr 26 12:26:06 2021 [3132/49] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/50] * debug/client	 Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/51] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/52] * debug/network	 Sent HEL message
Mon Apr 26 12:26:06 2021 [3132/53] * debug/client	 Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/54] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/55] * debug/network	 Received ACK message
Mon Apr 26 12:26:06 2021 [3132/56] * debug/client	 Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/57] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/58] * debug/channel	 Connection 988 | SecureChannel 0 | Generating new local nonce
Mon Apr 26 12:26:06 2021 [3132/59] * debug/channel	 Connection 988 | SecureChannel 0 | Requesting to open a SecureChannel
Mon Apr 26 12:26:06 2021 [3132/60] * debug/channel	 Requesting to open a SecureChannel
Mon Apr 26 12:26:06 2021 [3132/61] * debug/channel	 OPN message sent
Mon Apr 26 12:26:06 2021 [3132/62] * debug/client	 Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/63] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/64] * trace/channel	 Connection 988 | SecureChannel 38 | Verifying chunk signature
Mon Apr 26 12:26:06 2021 [3132/65] * trace/channel	 Connection 988 | SecureChannel 38 | Generating new local keys
Mon Apr 26 12:26:06 2021 [3132/66] * info/channel	 Connection 988 | SecureChannel 38 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
Mon Apr 26 12:26:06 2021 [3132/67] * info/client	 Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/68] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/69] * debug/channel	 Connection 988 | SecureChannel 38 | Sending request with RequestId 2 of type GetEndpointsRequest
Mon Apr 26 12:26:06 2021 [3132/113] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/114] * trace/channel	 Connection 988 | SecureChannel 38 | Generating new remote keys
Mon Apr 26 12:26:06 2021 [3132/115] * info/client	 Selected Endpoint opc.tcp://172.16.65.80:10001 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Mon Apr 26 12:26:06 2021 [3132/116] * info/client	 Selected UserTokenPolicy open62541-username-policy with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Mon Apr 26 12:26:06 2021 [3132/117] * info/client	 Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/118] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/119] * warn/client	 The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
Mon Apr 26 12:26:06 2021 [3132/120] * debug/client	 Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/121] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/122] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/123] * info/securitypolicy	 The basic256sha256 security policy channel with openssl is created.
Mon Apr 26 12:26:06 2021 [3132/124] * debug/network	 Sent HEL message
Mon Apr 26 12:26:06 2021 [3132/125] * debug/client	 Client Status: ChannelState: HELSent, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/126] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/127] * debug/network	 Received ACK message
Mon Apr 26 12:26:06 2021 [3132/128] * debug/client	 Client Status: ChannelState: AckReceived, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/129] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/130] * debug/channel	 Connection 1008 | SecureChannel 0 | Generating new local nonce
Mon Apr 26 12:26:06 2021 [3132/131] * debug/channel	 Connection 1008 | SecureChannel 0 | Requesting to open a SecureChannel
Mon Apr 26 12:26:06 2021 [3132/132] * debug/channel	 Requesting to open a SecureChannel
Mon Apr 26 12:26:06 2021 [3132/133] * debug/channel	 OPN message sent
Mon Apr 26 12:26:06 2021 [3132/134] * debug/client	 Client Status: ChannelState: OPNSent, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/135] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/136] * trace/channel	 Connection 1008 | SecureChannel 39 | Verifying chunk signature
Mon Apr 26 12:26:06 2021 [3132/137] * trace/channel	 Connection 1008 | SecureChannel 39 | Calculated padding size to be 76
Mon Apr 26 12:26:06 2021 [3132/138] * trace/channel	 Connection 1008 | SecureChannel 39 | Generating new local keys
Mon Apr 26 12:26:06 2021 [3132/139] * info/channel	 Connection 1008 | SecureChannel 39 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
Mon Apr 26 12:26:06 2021 [3132/140] * info/client	 Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
Mon Apr 26 12:26:06 2021 [3132/141] * trace/client	 Client connect iterate
Mon Apr 26 12:26:06 2021 [3132/142] * debug/channel	 Connection 1008 | SecureChannel 39 | Sending request with RequestId 5 of type CreateSessionRequest
```

However, at this point, the thread of execution dies, and no status code is returned.

Because I'm trying to connect to the sample certified server (server_ctt), I also have a similar log from the server as well (but regardless, I believe that it is on the client side where the problem is).

[... 68 lines cut ...]
    symEncryptionAlgorithm->uri = UA_STRING("http://www.w3.org/2001/04/xmlenc#aes256-cbc\0");        
    symEncryptionAlgorithm->getLocalKeyLength = UA_SymEn_Basic256Sha256_getLocalKeyLength;
    symEncryptionAlgorithm->getRemoteKeyLength = UA_SymEn_Basic256Sha256_getRemoteKeyLength;
    symEncryptionAlgorithm->getRemoteBlockSize = UA_SymEn_Basic256Sha256_getRemoteBlockSize;
    symEncryptionAlgorithm->decrypt = UA_SymEn_Basic256Sha256_decrypt;
    symEncryptionAlgorithm->encrypt = UA_SymEn_Basic256Sha256_encrypt;
```

Notably, in this section of code, many of the same attributes are set for each of asymmetric and symmetric algorithms. However, in the case of the symmetric algorithm, getRemotePlainTextBlockSize remains unset, where as it is set to UA_AsymEn_Basic256Sha256_getRemotePlainTextBlockSize for the asymmetric algorithm.

Might this be the cause of the problem that I'm having?

At the moment, I have no explanation as to why my code seems to work on Mac/Linux when I've been having these issues on Windows. It seems to me that an issue like this one, if it were to cause a problem on one OS, would probably also cause problems on other OSs as well. So far, I have only encountered this problem on Windows, and very similar code in my program has been able to otherwise connect to the server_ctt sample server without any problems.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): version 1.2
 - [x] Other OPC UA SDKs used (client or server): attempting to connect to server_ctt
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-27:
Yes. Whether this triggered your particular issue or not, the callback was missing for OpenSSL.
This was not an issue before, but the assert checks for the block sizes to match.
Added via #4373.

ixfg9922, 2021-04-28:
Awesome. Thank you.

closed 2021-04-28 by ixfg9922