open62541/open62541 #4372: Upcoming mbedtls V3 contains breaking changes to API

GoetzGoerisch, 2021-04-27:
## Description

With mbedtls moving to V3 they are unifying the internal headers into library/* which breaks current implementations.
see https://github.com/ARMmbed/mbedtls/pull/4164

## Background Information / Reproduction Steps
The [sample server](https://github.com/umati/Sample-Server) uses mbedtls with automated [dependency updates](https://github.com/umati/Sample-Server/pull/222).
The latest changes from ARMmbed/mbedtls@b5939e8 to ARMmbed/mbedtls@12f93f4. breaks the CI, as it upgrades from V2 to V3. 
In detail it is this commit ARMmbed/mbedtls@ea0a865

```
 Move entropy_poll.h to library

`entropy_poll.h` is not supposed to be used by application code and
is therefore being made internal.

```

Used CMake options:

```bash
cmake -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON -DUA_NAMESPACE_ZERO:STRING=FULL -DUA_ENABLE_ENCRYPTION:BOOL=1 -DUA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=1 

```
https://github.com/umati/Sample-Server/blob/2d282e64dcccaf0b2591b6df46436e3c5785a37a/.github/CMakeLists.txt#L28

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0c2cb1c
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-04-27:
Two options. 

1. We provide code that works for v2 and v3.
Potentially with a compiler macro that comes from a standard header.

2. Fix mbedtls to a long-term support branch.
That should be available as a lot of software has that same problem.

GoetzGoerisch, 2021-04-28:
Thanks,

would prefer solution 1. 
I'm going to look into it.

GoetzGoerisch, 2021-05-05:
For reference: https://github.com/ARMmbed/mbedtls/blob/development/docs/3.0-migration-guide.md#internal--alt-focused-headers-were-moved-to-a-private-location

> If you're a library user and used to rely on having access to a structure or function that's now in a private header, please reach out on the mailing list and explain your need; we'll consider adding a new API in a future version.

GoetzGoerisch, 2022-05-02:
Fixed with #5071.
Thank you!

closed 2022-05-02 by GoetzGoerisch