open62541/open62541 #4374: Zero tolerance on event loss, server implementation

marktomm, 2021-04-28:
## Description

I am receiving events for data-points (from lower level devices) and storing them in database until they are transferred to uplink (i.e. SCADA). I am trying to implement OPC UA Server to send the events stored in db to OPC UA SCADA Client. It is necessary that all the events from db reach the Client.

Only one client is connected to the server.

I am creating a Variable in OPC UA AddressSpace per each data-point. Each data-point may have a history of events in db. My idea is to attach a UA_DataSource read callback to each Variable Value and get the oldest data-point event from db every time the OPC UA Client makes a read request. The idea is that Client subscribes to the Variable Value and recvs all the events that are in the db and then all the next ones as they appear in db.

## Background Information / Reproduction Steps

I am having multiple issues with the described approach. The below logs are from the example repo.

### Events lost on session activate

```text
$ ./server # start the server. The server has 10 Int32 events in db: 9,8,7,...,0
[2021-04-28 09:10:47.658 (UTC+0300)] info/server        event buffer size set to 10
[2021-04-28 09:10:47.661 (UTC+0300)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2021-04-28 09:10:47.661 (UTC+0300)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2021-04-28 09:10:47.661 (UTC+0300)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-04-28 09:10:47.661 (UTC+0300)] info/server        send value: 9, session Id: g=00000001-0000-0000-0000-000000000000
[2021-04-28 09:10:47.661 (UTC+0300)] info/server        send value: 8, session Id: g=00000001-0000-0000-0000-000000000000
[2021-04-28 09:10:49.773 (UTC+0300)] info/server        Activate session ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:49.773 (UTC+0300)] info/server        send value: 7, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:49.773 (UTC+0300)] info/server        send value: 6, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:50.024 (UTC+0300)] info/server        send value: 5, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:50.274 (UTC+0300)] info/server        send value: 4, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:50.524 (UTC+0300)] info/server        send value: 3, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:50.774 (UTC+0300)] info/server        send value: 2, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:51.024 (UTC+0300)] info/server        send value: 1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:51.274 (UTC+0300)] info/server        send value: 0, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:51.524 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:51.774 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:52.024 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:52.274 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:52.524 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:52.774 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:53.024 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
[2021-04-28 09:10:53.072 (UTC+0300)] info/server        Close session ns=1;g=4f635bab-ee32-f419-07f0-7ff79df96f56
^C[2021-04-28 09:10:54.204 (UTC+0300)] info/server      received ctrl-c
```

```text
$ ./client # start the client in a separate terminal while the server is running. Subscribe to Variable Value
[2021-04-28 09:10:49.772 (UTC+0300)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-04-28 09:10:49.773 (UTC+0300)] info/userland      Create subscription succeeded, id 1
[2021-04-28 09:10:49.773 (UTC+0300)] info/userland      Monitoring 1, 1
[2021-04-28 09:10:50.274 (UTC+0300)] info/userland      Got value: 6
[2021-04-28 09:10:50.274 (UTC+0300)] info/userland      Got value: 5
[2021-04-28 09:10:50.774 (UTC+0300)] info/userland      Got value: 4
[2021-04-28 09:10:50.774 (UTC+0300)] info/userland      Got value: 3
[2021-04-28 09:10:51.274 (UTC+0300)] info/userland      Got value: 2
[2021-04-28 09:10:51.274 (UTC+0300)] info/userland      Got value: 1
[2021-04-28 09:10:51.774 (UTC+0300)] info/userland      Got value: 0
[2021-04-28 09:10:51.774 (UTC+0300)] info/userland      Got value: -1
^C[2021-04-28 09:10:53.071 (UTC+0300)] info/userland    received ctrl-c
[2021-04-28 09:10:53.072 (UTC+0300)] warn/channel       Connection 3 | SecureChannel 1 | Receiving the response failed with StatusCode BadSecurityChecksFailed
[2021-04-28 09:10:53.072 (UTC+0300)] error/userland     Subscription Id 1 was deleted
```

First two events 9, 8 fly off within some internal session prior to any client connection. A sub-optimal solution is checking session id prior to popping anything from db. 

Another issue is that after activating a session, server always runs read handler within client session context and the first read value, in our case 7, never reaches the client. pcap supplied `client-server-1.pcapng`. It is not sent out by the server.

### Events go nowhere on client abrupt disconnect

```text
$ ./server
[2021-04-28 09:18:26.649 (UTC+0300)] info/server        event buffer size set to 10
[2021-04-28 09:18:26.651 (UTC+0300)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[... 58 lines cut ...]
Used CMake options:

```bash
git clone https://github.com/open62541/open62541.git
cd open62541
git co v1.2.1 # commit 4038a318952e3d7d77a85f205d279cdfabc81930
git submodule update --init --recursive 
mkdir build 
cd build
cmake -DUA_LOGLEVEL=100 -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_FULL_NS0=1 -DUA_DEBUG=1 ..
make -j9
sudo make install
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.1 commit 4038a318952e3d7d77a85f205d279cdfabc81930
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux mark-pc-manjaro 5.10.26-1-MANJARO x86_64 GNU/Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached: [wireshark.tar.gz](https://github.com/open62541/open62541/files/6389710/wireshark.tar.gz)
 - [x] Self-contained code example attached: https://github.com/marktomm/open62541-buffer-datasource-sub
 - [ ] Critical issue

jpfr, 2021-05-05:
OPC UA MonitoredItems come in two versions.

1. Data-MonitoredItems that listen on changes in the information model with a sampling interval.
2. Event-MonitoredItems that listen on discrete events generated by an object in the information model.
Only objects can generate events. Variables cannot.

We have a few code examples for events you could take a look at.
Furthermore the standard defines a Service for access to historical events.
But support is still limited in open62541.

marktomm, 2021-05-14:
## Data-MonitoredItems

Did more tests with Data-MonitoredItems

### Observations using datasource approach:
```
$ ./server | grep 'send value'
[2021-05-04 08:25:43.499 (UTC+0300)] info/server        send value: 9, session Id: g=00000001-0000-0000-0000-000000000000
[2021-05-04 08:25:43.499 (UTC+0300)] info/server        send value: 8, session Id: g=00000001-0000-0000-0000-000000000000
[2021-05-04 08:25:53.235 (UTC+0300)] info/server        send value: 7, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:53.235 (UTC+0300)] info/server        send value: 6, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:53.486 (UTC+0300)] info/server        send value: 5, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:53.736 (UTC+0300)] info/server        send value: 4, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:53.986 (UTC+0300)] info/server        send value: 3, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:54.236 (UTC+0300)] info/server        send value: 2, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:54.486 (UTC+0300)] info/server        send value: 1, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:54.736 (UTC+0300)] info/server        send value: 0, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8
[2021-05-04 08:25:54.986 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=87a1d9f6-e7fb-9556-20b2-7b377cc78cd8

$ ./client | grep 'Got value'
[2021-05-04 08:25:53.736 (UTC+0300)] info/userland      Got value: 6
[2021-05-04 08:25:53.736 (UTC+0300)] info/userland      Got value: 5
[2021-05-04 08:25:54.236 (UTC+0300)] info/userland      Got value: 4
[2021-05-04 08:25:54.236 (UTC+0300)] info/userland      Got value: 3
[2021-05-04 08:25:54.736 (UTC+0300)] info/userland      Got value: 2
[2021-05-04 08:25:54.736 (UTC+0300)] info/userland      Got value: 1
[2021-05-04 08:25:55.235 (UTC+0300)] info/userland      Got value: 0
[2021-05-04 08:25:55.235 (UTC+0300)] info/userland      Got value: -1
```
1. the callback gets called twice on instantiation/internally, so events 9 and 8 do not reach the client
1. within client session the first callback value sent out to by the server does not reach the wire, in this case event 7

### Observations with value callback:

```
$ ./server | grep 'send value'
[2021-05-04 08:30:09.952 (UTC+0300)] info/server        send value: 9, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:09.952 (UTC+0300)] info/server        send value: 8, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:10.202 (UTC+0300)] info/server        send value: 7, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:10.452 (UTC+0300)] info/server        send value: 6, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:10.703 (UTC+0300)] info/server        send value: 5, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:10.952 (UTC+0300)] info/server        send value: 4, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:11.202 (UTC+0300)] info/server        send value: 3, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:11.452 (UTC+0300)] info/server        send value: 2, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:11.702 (UTC+0300)] info/server        send value: 1, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:11.953 (UTC+0300)] info/server        send value: 0, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0
[2021-05-04 08:30:12.202 (UTC+0300)] info/server        send value: -1, session Id: ns=1;g=cb507801-d66f-8fa5-7d17-810845a40ac0

$ ./client | grep 'Got value'
[2021-05-04 08:30:10.452 (UTC+0300)] info/userland      Got value: 8
[2021-05-04 08:30:10.452 (UTC+0300)] info/userland      Got value: 7
[2021-05-04 08:30:10.952 (UTC+0300)] info/userland      Got value: 6
[2021-05-04 08:30:10.952 (UTC+0300)] info/userland      Got value: 5
[2021-05-04 08:30:11.453 (UTC+0300)] info/userland      Got value: 4
[2021-05-04 08:30:11.453 (UTC+0300)] info/userland      Got value: 3
[2021-05-04 08:30:11.953 (UTC+0300)] info/userland      Got value: 2
[2021-05-04 08:30:11.953 (UTC+0300)] info/userland      Got value: 1
[2021-05-04 08:30:12.452 (UTC+0300)] info/userland      Got value: 0
[2021-05-04 08:30:12.452 (UTC+0300)] info/userland      Got value: -1
```

1. within client session the first value sent out to by the server does not reach the wire

### Client recreating sub for some reason

server has value cb. cleint recreates sub and monitem on value 5 recvd
```
$ ./server | grep 'send value'
[2021-05-04 08:50:10.662 (UTC+0300)] info/server        send value: 9, session Id: ns=1;g=974dcb84-eb89-b5fa-e5be-2bf29789c88c
[2021-05-04 08:50:10.662 (UTC+0300)] info/server        send value: 8, session Id: ns=1;g=974dcb84-eb89-b5fa-e5be-2bf29789c88c
[... 18 lines cut ...]
[2021-05-04 08:50:11.663 (UTC+0300)] info/userland      Got value: 5
[2021-05-04 08:50:12.164 (UTC+0300)] info/userland      Got value: 2
[2021-05-04 08:50:12.164 (UTC+0300)] info/userland      Got value: 1
[2021-05-04 08:50:12.665 (UTC+0300)] info/userland      Got value: -1
```

On sub recreation events 4 and 3 go missing.

[... cut ...]
In addition while popping events form buffers on every read callback per Variable/Object/Property it is basically impossible to solve having multiple clients connect to one server at arbitrary times.
 
## Event based approach

I haven't tried Event based approach. 

## Historical events

This Service addresses all the requirements and allows multiple clients to have all the history.

Data backend example here: https://github.com/marktomm/open62541-buffer-history

Many thanks to @nicolasr75 for providing an example for sqlite: https://github.com/nicolasr75/open62541_sqlite in #2396

closed 2021-05-14 by marktomm