open62541/open62541 #4377: open62541 gcc Server compilation time

Psoke, 2021-04-29:
## Description
I am using Free OPC UA Modeler to design a custom Nodeset, compile it using the open62541 Nodeset Compiler and then use gcc to compile everything to a server. My open62541 was build with no shared libraries and no amalgamation. 

I am using this command to compile the custom nodeset:

`python3 ~/open62541/tools/nodeset_compiler/nodeset_compiler.py 
-–types-array=UA_TYPES 
-–existing ~/open62541/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml 
-–xml ~/Server/CustomNodeSet2.xml CustomNodeSet2
`

Works fine as far as I can judge. Then I use this command:

`gcc -std=c99 -I/home/ubuntu/install/include -L/home/ubuntu/install/lib CustomServer.c CustomNodeSet2.c -lopen62541 -o 
CustomServer
`

My CustomServer.c has the following code:

#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <signal.h>
#include <stdlib.h>
#include "CustomNodeSet2.h"
UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}
int main(int argc, char **argv) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);
    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    UA_StatusCode retval = CustomNodeSet2(server);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Could not add the example nodeset. Check previous output for any error.");
        retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    } else {
       retval = UA_Server_run(server, &running);
    }
    UA_Server_delete(server);
    return (int) retval;
}

The server is compiled correctly and currently running, can be browsed by UaXpert, everythin fine so far. But the gcc compile command takes more than 15 minutes to compile and is giving an error message at the end:

/home/alex/open62541/build/src_generated/open62541/namespace0_generated.c: In function ‘namespace0_generated’:
/home/alex/open62541/build/src_generated/open62541/namespace0_generated.c:118113:15: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
118113 | UA_StatusCode namespace0_generated(UA_Server *server) {
       |     

I run ubuntu on a VM but allocated it 4 cores and 4GB RAM and everything else is smooth and fast as expected. I searched for this problem but no solution so far. I am experimenting with designing custom information models and its very time consuming if every compilation takes 15 minutes and open62541 does not support runtime import of xml nodesets as far as I know (correct me?).
Thx for your time
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu VM on Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

r-si, 2021-04-30:
My compilation with gcc 10 on Windows (using TI CCS for am3359 embedded platform, using amalgamation, using full namespace zero) takes about 15 minutes, too. But finaly succeedes (one second before I lose patience). Build for Windows takes half a minute.

Psoke, 2021-04-30:
Mhm but why is that though? I am not that into compiling or platform-dependent machinecode but you can't tell me that a compilation of a ~20MB executable takes 15 minutes in 2021. Is gcc just poorly optimized or what is the matter here? 
On one youtube tutorial the compilation process takes only a couple of seconds, my error message is also not showing up there. Can't it be that gcc is running in some kind of endless loop with open62541?

r-si, 2021-04-30:
No idea. Assumedly its the resulting size of the source file. Using reduced namespace zero (which leads to a much smaller open62541.c), compilation time is ok.

cmbahadir, 2021-05-08:
Well, it seems that you have at least 118113 lines of generated namespace file but if the same file compiles faster in Ubuntu on the same PC (in a VM or in a container), then Windows Defender could be the reason for slowness.

Here is a stackoverflow post describing how to do exclude your source directories from searching:

https://stackoverflow.com/a/58433724

I am not sure if it helps and it may not be the safest solution, but it is worth trying :smile_cat:

jackybek, 2021-07-19:
I was running on raspberry pi 3 with 1gb ram and it took me 15-20 minutes to build the open62541 binaries with full namespace.
After I switch to raspberry pi 4 with 8 Gu ram, it only takes 10 minutes.
Yah more memory does helps.

andreasebner, 2025-07-30:
The issues seems to be solved since no more comments arrive. If you encounter such issues with a recent version feel free to create a new issue.

closed 2025-07-30 by andreasebner