open62541/open62541 #4378: pubsub Configuration file

audupi, 2021-04-29:
## Description
I am trying to implement OPC UA PubSub using **configuration file** method. In the example folder, I understand that `example_publisher.bin` file contains the configuration for setting up PubSub connection. 

Can you please share any information on how the **.bin** configuration file can be setup? 

## Background Information / Reproduction Steps

So far I have followed the open62541 examples,   

`./server_pubsub_file_configuration <portnumber> <path_to example_publisher.bin> `

Used CMake options:

```bash
cmake -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON ..
```

Fischer-Thomas, 2021-04-30:
Can you please provide the error logs you get when trying to use the example.

audupi, 2021-05-04:
Here, I have launched the `./server_pubsub_file_config` file in the terminal with the following command. The configuration file **example_publisher.bin** is in the folder path _~/open62541/examples/pubsub/example_publisher.bin_ . Hence the launch command

`./server_pubsub_file_configuration 4845 ~/open62541/examples/pubsub/example_publisher.bin`

I am able to connect UAExpert Client to the Server, when I shut the server down, I can see that the **PubSub** configuration is successful. All this is shown in the screen shot below. 

I would like to know mainly two things
1. I am not able to figure out what the pubsub_file_configuration is achieving? 
2. How do I subscribe to the Pubsub that is configured through the pubsub configuration file?
3. How to set up the `.bin` file, if I want to configure a PubSub on my Custom OPC UA Server. 

Thanks in advance!

[image]

Fischer-Thomas, 2021-05-05:
> 1. I am not able to figure out what the pubsub_file_configuration is achieving, as there is no change to the server variables as observed from the client.

The file configures a publisher on the server. You can find it when browsing the UAServer.
Root->Objects->Server->PublishSubscribe

As you correctly noticed there is no custom sampling over the variables and are only used by the Publisher to create a NetworkMessage.
You can either extend the code to have some sampling or change the variables using a client (e.g. UAExpert).

If you want to see the UADP-NetworkMessage i would suggest using wireshark.

> 2. How do I subscribe to the Publisher that is configured through the pubsub configuration file?

You can look at the configuration in the information model and configure a subscriber according to the publisher. Have a look at [tutorial_pubsub_subscribe.c](https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_subscribe.c) to undestand how to setup a subscriber.

> 3. How to set up the `.bin` file, if I want to configure a PubSub on my Custom OPC UA Server using a `.bin` file.

The content of the `.bin` file is an encoded UABinaryFileDataType which contains an object of the PubSubConfigurationDataType. As far as i know there is currently no tool available to create a PubSubConfiguration `.bin`.

closed 2021-05-17 by audupi