open62541/open62541 #4382: Creating a simple server and update his dataValues and variables

betodr4, 2021-04-30:
## Description
I am working in aproject with an ESP32 with the objective to create a OPCUA sever. The main idea is to define some variables and update their values every X seconds. I would like to know how to update variable values and which functions must I use. I have been studing the documentation and I have not seen a simple example to do that. 

I have been trying some example with functions that apears in the documentation, but I have not a clear idea.

Is there any repository with a simple example to look?

# Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): v1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: ESP32.
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

serega1103, 2022-06-10:
```
// SHORT INT
// Variable value from DB (for example)
short int* temperatureValue = (short int*)87;

UA_Int16 newValue = *temperatureValue;
UA_Variant newVariant;
UA_Variant_setScalar(&newVariant, &newValue, &UA_TYPES[UA_TYPES_INT16]);
UA_Server_writeValue(server, myNodeId, newVariant);

// INTEGER
// Variable value from DB (for example)
int* temperatureValue = (int*)51;

UA_Int32 newValue = *temperatureValue;
UA_Variant newVariant;
UA_Variant_setScalar(&newVariant, &newValue, &UA_TYPES[UA_TYPES_INT32]);
UA_Server_writeValue(server, myNodeId, newVariant);

// FLOAT
// Variable value from DB (for example)
float temperatureValue = 36.6;

UA_Float newValue = temperatureValue;
UA_Variant newVariant;
UA_Variant_setScalar(&newVariant, &newValue, &UA_TYPES[UA_TYPES_FLOAT]);
UA_Server_writeValue(server, myNodeId, newVariant);

// STRING
// Variable value from DB (for example)
char* colorFromCamera = (char*)"BLUE";

UA_String newValue;
newValue.length = strlen(colorFromCamera);
newValue.data = (UA_Byte*)colorFromCamera;
UA_Variant newVariant;
UA_Variant_setScalar(&newVariant, &newValue, &UA_TYPES[UA_TYPES_STRING]);
UA_Server_writeValue(server, myNodeId, newVariant);
```

andreasebner, 2025-07-30:
Closing due to inactivity.

closed 2025-07-30 by andreasebner