open62541/open62541 #4385: Custom Types not view in Codesys-OPCUA-client

robschw, 2021-05-02:
## Custom Types not view in Codesys-OPCUA-client
I tested the “server tested custom” with the latest open62541.c/h (V1.2;  9th Feb., 2021; program is running on VM with Ubuntu 64bit and eclipse) . It works fine with the software Unified Automation UaExpert (Version 1.4.6 237). All Node could be read and write. It seems to be OK.
But if I check the nodes with Codesys Application OPC UA Client (https://store.codesys.com/opc-ua-client-example.html ) the nodes hast no “sub nodes” for view and edit value.
Is there any special setting necessary for showing custom data/vartypes for using in the “Codesys-World”?

robschw, 2021-05-12:
Sorry my delay.
This issue needs a support by CoDeSys. 
I hope I could give you an answer.

paguss, 2022-08-04:
I have the same issue with CoDeSys OPCUA Client.
Do you have any update?
Thanks in advance.

andreasebner, 2025-07-30:
This looks like it's an issue with the Codesys Client. Since there are no replys i'am closing the issue. Since the codebase developed a lot, the issue is may fixed with recent versions.

closed 2025-07-30 by andreasebner