open62541/open62541 #4386: HEL/ ACK error during handshaking between opc server and rabbitmq

jackybek, 2021-05-03:
## Description
When I try to connect my opcserver to a rabbitmq, it returns HEL/ ACK error.
However, if I use mqtt, it can go through.

I checked the opcua specifications and understand that opcua handshake start with sending a HELlo and waiting for a ACKnowledgement.

when I check the source for enabling mqtt pubsub, I don’t see this instruction.  If someone would point me to the exact function that initiates this handshake by opcua server with mqtt server, I can then check if this is done properly for the handshake by Opcua server with rabbitmq server.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-05-05:
> When I try to connect my opcserver to a rabbitmq, it returns HEL/ ACK error.

There is no HEL message defined in the MQTT specification.
Are you trying to connect rabbitmq directly to an OPC UA server?
That cannot work...

jackybek, 2021-05-05:
Actually, it is the reverse.

I am trying to initiate a connection from opcua server to rabbitmq, just like what I did for the mqtt broker.

Open62541 server (app zone) -> rabbitmq broker (dmz)

A working instance : 
Open62541 server (app zone) -> mqtt broker (dmz)

jpfr, 2021-05-06:
> However, if I use mqtt, it can go through.

RabbitMQ is for AMQP. We don't have AMQP in the master branch.
How are you doing this?

jackybek, 2021-05-06:
I followed #3850 and installed proton lib for amqp