open62541/open62541 #4387: Question about the objectId argument in UA_Client_call

jurihock, 2021-05-03:
## Question

Both `UA_Client_readValueAttribute` and `UA_Client_writeValueAttribute` requires just a `nodeId` of the node to be accessed. But `UA_Client_call` requires an additional `objectId` of its parent node.

It looks a little odd to me to call `UA_Client_readValueAttribute(..., nodeId, ...)` but `UA_Client_call(..., objectId, nodeId, ...)`. I think, the `nodeId` should be meaningful enough in both cases, doesn't it?

Is there any chance to call a method identified just by the `nodeId` without the explicite knowledge of its `objectId`?

Thanks in advance!

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2021-05-04:
As you can use the same method node under multiple objects in a Server the server may require it:
```
Obj1(ns=2;i=20)->Method(ns=2;i=10)
Obj2(ns=2;i=30)->Method(ns=2;i=10)
```

jurihock, 2021-05-04:
Thanks for the hint!

The ability to share the equal method id across different objects indeed requires an additional argument...

So I'll close this issue.

closed 2021-05-04 by jurihock