open62541/open62541 #4388: UA_PUBSUB-How to print out received time data in subscriber side

Ozguryldz, 2021-05-03:
## Description
Hi All;
I'm quite new to open62541.I read about the structure for a month.  I just want to use open62541/examples/pubsub/tutorial_pubsub_publish.c and open62541/examples/pubsub/tutorial_pubsub_subscribe.c functions to print the time info that is intended to be send by publisher in every 100 ms. There is no exact application for this easy step in tutorials. OK in official webside, tutorials provides such a good starter for me but, until it comes to pubsub section. In pubsub tutorials, it says you can check your publisher works or not on wireshark. Why? Can't you give one or two more line of codes to print out what publisher sends? Or can't you give some printf commands to print out what subscriber gets?
tutorial_pubsub_subscribe.c function does not include any UA_LOG_INFO or printf method to print incoming data. So my problem is about reaching the actual sent data in subscriber side.

I have tried to give every detail about issue, but i think for this starter step my problem is very obvious for more experienced colligues.

This is all i have;
Terminal Log for tutorial_pubsub_publisher.c execute command; https://paste2.org/9YVmhM8P
Terminal Log for tutorial_pubsub_subscribe.c execute command;https://paste2.org/mPF6P5x9

There is nothing. I can see publisher send out some data, i can check it UDP connections in wireshark. But what is the data? I cannot reach to that "time" data. 

I used tutorial_pubsub_standalone.c function too which is shown in here;https://paste2.org/O1ksykhJ
but line 50 (retval = psc->receive(psc, &buffer, NULL, 100); ) always makes buffer size 0. And it causes the rest of the code not to be functional.(Where i can see some printf commands :) )

This issue should be very basic. But even in the official document of open62541.org PUBSUB /subcribe side was not explained in detail. 

Can someone help me please about : How should i update tutorial_pubsub_subscribe.c file to printout date and time info to the terminal?

## Background Information / Reproduction Steps

Used CMake options:

Include all CMake options here, which you modified or used for your build.
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=/usr/bin/bibtex
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/home/osboxes/install
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_TBZ2:BOOL=ON
CPACK_SOURCE_TGZ:BOOL=ON
CPACK_SOURCE_TXZ:BOOL=ON
CPACK_SOURCE_TZ:BOOL=ON
CPACK_SOURCE_ZIP:BOOL=OFF
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdfm
DVIPS_CONVERTER:FILEPATH=/usr/bin/dvips
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=/usr/bin/htlatex
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=/usr/bin/latex
LIB_BPF:FILEPATH=OFF
LUALATEX_COMPILER:FILEPATH=/usr/bin/lualatex
MAKEINDEX_COMPILER:FILEPATH=/usr/bin/makeindex
PDFLATEX_COMPILER:FILEPATH=/usr/bin/pdflatex
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
SPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-build
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ARCHITECTURE:STRING=posix
UA_ARCH_ADD_FLAGS:STRING=
UA_ARCH_EXTRA_INCLUDES:STRING=
UA_ARCH_LINKER_FLAGS:STRING=
UA_ARCH_REMOVE_FLAGS:STRING=
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_BUFMALLOC:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_ENCRYPTION:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=ON
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_PUBSUB_MONITORING:BOOL=ON
UA_ENABLE_PUBSUB_MQTT:BOOL=OFF
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=/usr/bin/xelatex
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):v1.2.1
 - [ ] Other OPC UA SDKs used (client or server): YES
 - [ ] Operating system:UBUNTU 16.04 LTE
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

audupi, 2021-05-03:
The values published in **tutorial_pubsub_publish.c** is received by the PubSub subscriber and updated in its server.  

Upon launching the **tutorial_pubsub_subscribe.c** example, note that a new OPC UA **server** is created. To visualize the values received by the PubSub Subscriber, simply connect a UAExpert Client to it.

Ozguryldz, 2021-05-03:
Hi audupi, thanks for reply. But AUExpert Client is not a method or function in open62541 library. It seems it is  program. Why should i need a program to get some easy data. It is just time data.

audupi, 2021-05-03:
You can also write a simple OPC UA client yourself if you do not want to use the UAExpert Client.

keep in mind, the **tutorial_pubsub_subscriber.c** launches a server upon launch and updates the values received from the publisher. 

Hope this helps.

Ozguryldz, 2021-05-03:
@audupi  Thanks for this explanation!!
Yes i could get the published message data with the client application for one time. 
Now i need to make it embedded to tutorial_pubsub_subscribe.c file with one additional listener API. 
Then it should be printed out on every time subscriber gets message(every 100 microsecond)
I"m really stranger to git environment. I don"t know even how we should close an issue. But i will close this one. 
Best Regards
Ozgur

closed 2021-05-05 by Ozguryldz