open62541/open62541 #4389: Server stuck due to a misbehaving client

EmmanuelP, 2021-05-04:
## Description

I'm trying to understand a behaviour I'm seeing on a setup with one server, and 2 clients.

It is a very simple server, with one node. Each client open a subscription and monitor this node.If I suspend one of the client (ctrl+z), the server CPU usage goes to 100% and the other client does not receive data anymore.

I'm a bit surprised that a misbehaving client can so easily stop a server.

To help you reproduce this issue, I have created a repository with all the relevant informations here: https://github.com/EmmanuelP/opctest

## Background Information / Reproduction Steps

Used CMake options: `-DBUILD_SHARED_LIBS=ON -DBUILD_STATIC_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=REDUCED -DOPEN62541_VERSION=v1.2 -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON -DUA_ARCH_REMOVE_FLAGS="-Werror" -DUA_MULTITHREADING=200`

```
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
CMAKE_INSTALL_PREFIX:PATH=/home/pacaud/virgo/App/OpcTest/git/Linux-x86_64-FED32
MDNSD_LOGLEVEL:STRING=300
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=ON
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=200
UA_NAMESPACE_ZERO:STRING=REDUCED
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Fedora 34
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: see README.md  of opctest repository
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

EmmanuelP, 2021-05-04:
I'm able to reproduce the issue on ubuntu 20.10 (in a VM running on a Fedora 34 host), using ppa from open6251-team (libopen62541-1-dev/groovy 20210128T143534\~1.1.5\~1\~ge1650d8a\~ubuntu20.10.1).

[image]

EmmanuelP, 2021-05-04:
Reproducible also on a CentOS 7 system:

[image]

jpfr, 2021-05-05:
Tested with v1.2 and master. The error does not occur there.
Will test this with 1.1 as well.

A fix to get the ppa up-to-date is on the way.

EmmanuelP, 2021-05-05:
Hi Julius,

Thanks for your help. Could you tell me what OS version you are using ? I will try to replicate your setup.

Did you actually just stop the consumer process, not kill it ?

EmmanuelP, 2021-12-02:
The 100% CPU use should be fixed by #4704 .

Still, the main issue is still there, connection_write will continue forever unless the client is unstopped (by issuing `fg` for example), or if the connection breaks.

I guess the same situation can be triggered if the client is not able to receive quickly enough the incoming data. At some point the output socket buffer of the server will be full, and connection_write will continue as long as the buffer is full, preventing the other clients to receive data. I have not checked this assumption yet.

andreasebner, 2025-07-30:
Thanks for reporting. Versions 1.4 is recent now, with 1.5 upcomming. If problems in theses versions persist, feel free to reopen.

closed 2025-07-30 by andreasebner