open62541/open62541 #4390: Unable to use aa_tree API when building as shared lib

cldhms, 2021-05-04:
## Description
I'm trying to use the aa_tree API in my own code when building the stack as a shared lib but I'm not able to due to linking errors. The problem seems to be the absence of the UA_EXPORT macro for the function declarations in the header file. Is it possible to solve my issue in any other way or do you wan't me to create a PR with my proposed fix? I assume I shouldn't use the UA_EXPORT macro though as the aa_tree implementation is independent from the open62541 stack, or is this acceptable?

Used CMake options:

-DBUILD_SHARED_LIBS=ON

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.2.1
 - [X] Other OPC UA SDKs used (client or server): N/A
 - [X] Operating system: N/A
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-05-05:
The aa_tree structure is internal and not exposed in the public headers.
The code is very much stable. Wouldn't it work for you to just copy aa_tree.c/h into your project and static-compile it?

cldhms, 2021-05-05:
I'm not too familiar with the folder structure here but I was under the impression that it was supposed to be a public API. aa_tree.h is among the exported_headers in cmake and is exported along with ms_stdint.h and ziptree.h and the open62541 folder for me. Is only the content of the open62541 include folder supposed to be public?

Copying the files would require me to change the namespace for the copied files as I'm already able to include aa_tree.h, but that's also a possibility if needed.

o6-nbeier, 2026-10-20:
Thank you for the question.

The AA-tree is a private container of open62541 and is not exported from the shared library (it was removed on newer branches), so compile your own copy into your application.

Closing for now.

closed 2026-10-20 by o6-nbeier