open62541/open62541 #4394: why skipType(name) in nodeset_compiler/type_parser.py ignore the data type name end with 'NodeId'

reidjunli123, 2021-05-06:
why skipType(name) in nodeset_compiler/type_parser.py ignore the node name end with 'NodeId', it means when the custom data type name end with 'NodeId' will not be created, any special consideration for it? It seems that a limit for the custom data type name.

both
the master branch of https://github.com/open62541/open62541/blob/master/tools/nodeset_compiler/type_parser.py
release 1.2 branch of https://github.com/open62541/open62541/blob/1.2/tools/nodeset_compiler/type_parser.py
have this issue.

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-05-06:
Fixed via #4396.

reidjunli123, 2021-05-07:
thanks for quick update

closed 2021-05-07 by reidjunli123