open62541/open62541 #4399: Pubsub /On Subcriber side Up on subscribed message received, Buffer length becomes immediately 0

Ozguryldz, 2021-05-07:
## Description
Hi All ;

I'm trying to add multiple data to my pubsub implementation. These datas are simple like double or UA_int32 type variables.
In Subscriber side, instead of creation of a new server(so by additional another client app i can get the data) i want to get the published data directly. For this implementation, as subscriber i need to use open62541/examples/pubsub/pubsub_subscribe_standalone.c file.

The problem is while i can get Date time info on subscriber side(I should be able to get it every seconds but somehow i mixed-up the SW now and i can get it on every ten seconds)

## Background Information / Reproduction Steps
About this tutorial level implementation, i have a few questions.
Used source files are:
tutorial_pubsub_publish.c -> publishes system time info and an int32 number =42 on every 1000 ms(Server port number 4840 and networkAddressUrl="opc.udp://224.0.0.22:4840/")
pubsub_subscribe_standalone.c->registers to "opc.udp://224.0.0.22:4840/" and receives data on every 1000  ms

For full code files:
https://github.com/Ozguryldz/PubSubTrials/blob/e5e19ee2605af8f56cb43c28d7f7493693614ca0/tutorial_pubsub_publish.c
https://github.com/Ozguryldz/PubSubTrials/blob/e5e19ee2605af8f56cb43c28d7f7493693614ca0/pubsub_subscribe_standalone.c

I can alrerady get time info. And By modifying 'tutorial_pubsub_publish.c' file i could add a new variable to the server with below additional SW line:

    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = 42;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.displayName = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    /* Add the variable node to the information model */

    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
    parentReferenceNodeId, myIntegerName,
    UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

and add this variable to the publishedDataSet by below code:
    addPubSubConnection(server, transportProfile, networkAddressUrl);
    addPublishedDataSet(server);
    addDataSetField(server);
    addVariableDataSetField(server, 1, "the.answer");//i have implemented addVariableDataSetField
    addWriterGroup(server);
    addDataSetWriter(server);

//i have implemented addVariableDataSetField as below by getting help from #2598 issue.
static void
addVariableDataSetField(UA_Server *server, int nsIndex, char* qualifier){
    /* Add a field to the previous created PublishedDataSet */
    UA_NodeId dataSetFieldIdent;
    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING("temperature");
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    //dataSetFieldConfig.field.variable.publishParameters.publishedVariable =
    //UA_NODEID_NUMERIC(nsIndex, numIdent);
	dataSetFieldConfig.field.variable.publishParameters.publishedVariable =
    UA_NODEID_STRING(nsIndex, qualifier);
    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, publishedDataSetIdent,
                              &dataSetFieldConfig, &dataSetFieldIdent);
}

Problem Description:
1)I can see added new variable in UA CLient after executing tutorial_pubsub_publish.c and its namespaceindex and its identifier are correct as i expected. So i push these variables to the addVariableDataSetField(UA_Server *server, int nsIndex, char* qualifier) function.
2)On subscriber side my log is as follow:https://paste2.org/ydEhk9NX

So i get 25 byte length message on every second and 44 byte length message on every 10 seconds. Date time info is received on every 10 seconds and it can be printed. 25 byte i suppose it is my additional variable in my PUBSUB connection.(I can also see the field count is 2) However i cannot print it to the terminal. 

Can anybody have an idea about the issue/issues over here?

Used CMake options:

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):(release number or git tag):v1.2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:UBUNTU 16.04 LTE
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached YES
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

cmbahadir, 2021-05-08:
Hello @Ozguryldz,

The type of published variable is `UA_Int32` in your publisher code but the type check is done for `UA_UInt32` in subscriber.

You have to change below  to print `UA_Int32`

https://github.com/Ozguryldz/PubSubTrials/blob/e5e19ee2605af8f56cb43c28d7f7493693614ca0/pubsub_subscribe_standalone.c#L108-L113

Ozguryldz, 2021-05-08:
Hello @cmbahadir ; 
Hi to all Istanbul and izmir;)

thanks for great explanation. Sometimes an eye from outside can realize an additional "U" letter which changes result entirely :)
Now i can print int32 type data too. 

publisher sends out these data on every seconds.(publishing interval is 1000 ms and receive interval is 1000 ms as well) 
However my SUbscriber log is still like in here:https://paste2.org/ydEhk9NX
Only difference now i can get int32 type data too. 

If i cancel publishing int32 type data, date&time info is printed on every seconds without an issue.

closed 2021-05-11 by Ozguryldz