open62541/open62541 #4400: Array of customer data type

robschw, 2021-05-09:
# Description
I added the sample “custom data type” with an array of point.
But by init variable, the occurs an error: BadTypeMismatch

my 
[array_of_customdatatype_test.zip](https://github.com/open62541/open62541/files/6447098/array_of_customdatatype_test.zip)

code:
```
static void
add3DPointVariableArray(UA_Server *server) {
	UA_StatusCode retval;
    int i;
	Point p[10];
	for (i=0;i<10;i++)
	{
		p[i].x = 3.0 + (UA_Float)((i+1)*100);
		p[i].y = 4.0 + (UA_Float)((i+1)*100);
		p[i].z = 5.0 + (UA_Float)((i+1)*100);
	}
    UA_VariableAttributes vattr = UA_VariableAttributes_default;
    vattr.description = UA_LOCALIZEDTEXT("en-US", "3D array");
    vattr.displayName = UA_LOCALIZEDTEXT("en-US", "3D array");
    vattr.dataType = PointType.typeId;
    vattr.valueRank = UA_VALUERANK_ONE_DIMENSION;
	UA_UInt32 arrayDims[1] = { 10 };
	vattr.arrayDimensions 	= arrayDims;
	vattr.arrayDimensionsSize = 1;
	UA_Variant_setArray(&vattr.value, p, 10, &PointType);

   	vattr.arrayDimensions = arrayDims;
   	vattr.arrayDimensionsSize = 1;
   	retval = UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "3D.Array"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                              UA_QUALIFIEDNAME(1, "3D.Array"),
                              pointVariableTypeId, vattr, NULL, NULL);
   	printf(">>>>> Add Var Array: %s\n", UA_StatusCode_name(retval));
}
```

## Background Information / Reproduction Steps
In machine controls it is sometimes necessary to use an array of customer-specific data structures to transfer data quickly. Therefore (I think) I is a very good ide to make also an example with array of data with custom data types.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.2-rc2-45-gf4270ceb
 - [X] Other OPC UA SDKs used (client or server):  UA expert
 - [X] Operating system: Ubuntu 20.4, eclipse
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-05-09:
Looks correct. Please show what’s behind pointVariableTypeId .

robschw, 2021-05-09:
Behind pointVariable.TypeId is following code (see [open62541/examples/custom_datatype/client_types_custom.c ](https://github.com/open62541/open62541/blob/master/examples/custom_datatype/client_types_custom.c))

`const UA_NodeId pointVariableTypeId = { 1, UA_NODEIDTYPE_NUMERIC, {4243}};`

robschw, 2021-05-12:
Sorry,
It was my mistake.
The following code works.
I think the parameter “`const UA_NodeId typeDefinition`” of is not the declare “`pointVariableTypeId`” 

declare in example [open62541/examples/custom_datatype/](https://github.com/open62541/open62541/tree/master/examples/custom_datatype)
``` const UA_NodeId pointVariableTypeId = {1, UA_NODEIDTYPE_NUMERIC, {4243}};), it is “UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE)”.  ```

So my code is:
``` 
static void add3DPointVariableArray(UA_Server *server) {
	UA_StatusCode retval;
    int i;
	Point p[10];
	for (i=0;i<10;i++)
	{
		p[i].x = 3.0 + (UA_Float)((i+1)*100);
		p[i].y = 4.0 + (UA_Float)((i+1)*100);
		p[i].z = 5.0 + (UA_Float)((i+1)*100);
	}
    UA_VariableAttributes vattr = UA_VariableAttributes_default;
    vattr.description = UA_LOCALIZEDTEXT("en-US", "3D array");
    vattr.displayName = UA_LOCALIZEDTEXT("en-US", "3D array");
    vattr.dataType = PointType.typeId;
    vattr.valueRank = UA_VALUERANK_ONE_DIMENSION;
	UA_UInt32 arrayDims[1] = { 10 };
	vattr.arrayDimensions 	= arrayDims;
	vattr.arrayDimensionsSize = 1;
	UA_Variant_setArray(&vattr.value, p, 10, &PointType);

   	vattr.arrayDimensions = arrayDims;
   	vattr.arrayDimensionsSize = 1;
   	retval = UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "3D.Point.Array"),
            					UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
								UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
								UA_QUALIFIEDNAME(1, "3D.Point.Array"),
								/* pointVariableTypeId, */
								UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
								vattr, NULL, NULL);
   	printf(">>>>> Add Var Array: %s\n", UA_StatusCode_name(retval));
} 
```

closed 2021-05-12 by robschw