open62541/open62541 #4409: Commit on Apr 11 (b13a387) breaks PubSub

rexsum, 2021-05-13:
## Description

Commit on Apr 11 (b13a387) breaks PubSub.

## Background Information / Reproduction Steps

A bug that was introduced in the Apr 11 commit that causes PubSub ethernet to fail, here's the details.

The following line:

        if(memcmp(eth_hdr.ether_dhost, channelDataEthernet->targetAddress, ETH_ALEN) != 0) {

was replaced by this code:

        const UA_Byte *compareAddress = is_multicast_address(channelDataEthernet->targetAddress) ? eth_hdr.ether_dhost : eth_hdr.ether_shost;
        if(memcmp(compareAddress, channelDataEthernet->targetAddress, ETH_ALEN) != 0) {

The problem is that eth_hdr.ether_shost is always: 00 00 00 00 00 00

This causes the memcmp to always not match resulting in the function returning with: UA_STATUSCODE_GOODNODATA 

I have had to patch the code to backout this commit and get my application working again.

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_SHARED_LIBS=ON -DUA_LOGLEVEL=300 \
      -DUA_MULTITHREADING=100 -DUA_ENABLE_IMMUTABLE_NODES=ON \
      -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON \
      -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON \
      -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_METHODCALLS=ON -DUA_NAMESPACE_ZERO=REDUCED \
      ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): latest master commit
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

MurzynJ, 2021-05-19:
@rexsum 
Sorry that my change caused you troubles. May I ask you some questions, which will help to understand the issue?

From the description of your issue I assume 2 things:
1) The address of your PubSub connection, represented by channelDataEthernet->targetAddress, is a unicast address.
Otherwise you would not run into the problem with eth_hdr.ether_shost, because the compareAddress would have been set to point on eth_hdr.ether_dhost in the previous line.
2) Actually the address of your PubSub connection, represented by channelDataEthernet->targetAddress, is your own MAC address.
Otherwise you would not be able to receive the frames with the original code, where the memcmp is comparing eth_hdr.ether_dhost (destination) with an address of the PubSub connection represented by channelDataEthernet->targetAddress.

If the second statement is correct, there is a problem with your PubSub configuration in my opinion. The PubSub connection's address in case of unicast, should be an address of the other end of the connection, not the local one.
Apart from that, it seems strange to me that eth_hdr.ether_shost (source) is always all 0s. Is it expected behaviour on Linux, do you know? I'm testing it on some real time system with a 3rd party network stack and I'm getting the actual source address from the received frame in the eth_hdr.ether_shost.

rexsum, 2021-05-24:
Thank you for responding. I was apparently too hasty in my original analysis. The eth_hdr.ether_shost value is not always hex zeros (I was looking at the publisher) but that's not actually relevant to the issue.

The real issue is that this PR fundamentally changes the way that subscribers need to specify the network url. This change requires that existing applications be recompiled or reconfigured.

The way it was working prior to the PR is that the subscriber always specified the MAC address of its own NIC card when receiving unicast broadcast or would specify hex FF's when receiving multicast.

This PR changes this so that the MAC address of the publisher must always be specified. This is true whether the broadcast is unicast or multicast. This change also precludes the subscriber from receiving broadcasts from multiple unicast publishers. This change will break some very valid use cases including my application.

The OPC Foundation specification Part 14 section 5.4.2 it describes how a subscriber can filter incoming messages using PublisherId, DataSetWriterId and DataSetClassId. It does not mention filtering based on the Publisher's address.

I performed much more testing this time around and how listed those results below:

Prior to PR
-----------
Publisher:
          network-url: "opc.eth://00-1f-c6-9c-8c-67"   # MAC address of subscriber's NIC
Subscriber:
          network-url: "opc.eth://00-1f-c6-9c-8c-67"   # MAC address of its own NIC

Publisher:
          network-url: "opc.eth://ff-ff-ff-ff-ff-ff"   # multicast
Subscriber:
          network-url: "opc.eth://ff-ff-ff-ff-ff-ff"   # multicast

After PR
--------
Publisher:
          network-url: "opc.eth://00-1f-c6-9c-8c-67"   # MAC address of subscriber's NIC
Subscriber:
          network-url: "opc.eth://00-1f-c6-9c-4d-31"   # MAC address of publisher's NIC

Publisher:
          network-url: "opc.eth://ff-ff-ff-ff-ff-ff"   # multicast
Subscriber:
          network-url: "opc.eth://00-1f-c6-9c-4d-31"   # MAC address of publisher's NIC

MurzynJ, 2021-06-08:
@rexsum Thank you for providing more details. Now I see exactly how my commit broke your application. Just one clarification, as I think that you mixed the following terms in your post:

- Unicast is ‘one to one’ communication, the destination MAC address of the frame has the least significant bit of a first octet (I/G bit) set to 0.
- Multicast is ‘one to many’ communication, the destination MAC address of the frame has the least significant bit of a first octet (I/G bit) set to 1.
- Broadcast is ‘one to all’ communication, the destination MAC address of the frame has all bytes set to FF.

Current version of PubSub specification (OPC UA Part 14 ver. 1.04) is not precise and allows for interpretations, which leads to interoperability problems in case of unicast communications. This issue has been feedback to the OPC UA WG and there will be changes implemented in the next specification release ver. 1.05.

Current implementation in open62541 assumes that on the unicast subscriber side, the PubSubConnection’s Address is set to subscriber’s own MAC, and on the publisher side it's also set to the subscriber's MAC. That’s how your application is configured. This works, but imagine you have a pair of devices, communicating bidirectionally, you would need to have 2 PubSubConnections on each side (one for publishing and one for subscribing), while logically, there is just one connection between 2 devices.

An alternative interpretation of PubSubConnection’s Address in case of unicast communication would be that it’s a MAC address of the communication partner, regardless on the publisher or on the subscriber side. Then, only 1 PubSubConnection is enough for both publishing and subscribing on each side of a logical connection. The change in the stack introduced by my commit, moved it to this second interpretation. For it to work, it is required to change the application configuration appropriately.

@jpfr 
Although both of the interpretations are equally correct (due to imprecise specification), and in my personal opinion the second interpretation is better, I recognize that I was too fast with proposing this change. Maybe it's better to wait for 1.05 to have a definitive instructions on how to interpret the PubSubConnection Address in any scenario (unicast vs. multicast/broadcast).

Additionally, thanks to @rexsum's test, I see that there is a bug in my implementation anyway, as it doesn’t take into account the broadcast communication, which should be handled the same as multicast. I’m doing a check using is_multicast_address(), while it should rather be opposite, I should be checking for ‘is unicast address’ and handle it differently than the rest i.e. multicast and broadcast.