open62541/open62541 #4414: example subscribe mqtt

davidem123, 2021-05-17:
I'm using  "tutorial_pubsub_mqtt_publish.c", but i can't find any example of "subscribe" with mqtt, why?how can i do subscribe over mqtt?

jackybek, 2021-07-11:
Wait for my companion documentation which is under review by Julius.

davidem123, 2021-07-13:
> Wait for my companion documentation which is under review by Julius.

where will it be available? will there be an example on github open62541/examples/pubsub/ ?

jackybek, 2021-07-13:
[OPCUAPubSubMQTT Documentation.txt](https://github.com/open62541/open62541/files/6810407/OPCUAPubSubMQTT.Documentation.txt)

davidem123, 2021-07-15:
> [OPCUAPubSubMQTT Documentation.txt](https://github.com/open62541/open62541/files/6810407/OPCUAPubSubMQTT.Documentation.txt)

thanks, but I'm looking for an example of subscribe_mqtt like tutorial_pubsub_mqtt_publish.c .I have 3 raspberrys where i runned:

davidem123, 2021-07-15:
> [OPCUAPubSubMQTT Documentation.txt](https://github.com/open62541/open62541/files/6810407/OPCUAPubSubMQTT.Documentation.txt)

thanks, but I'm looking for an example of subscribe_mqtt like tutorial_pubsub_mqtt_publish.c .I have 3 raspberrys where I want to run: 
-tutorial_pubsub_mqtt_publish.c
-mosquitto
-subscribe_mqtt ( I miss this part)

Tutorial_pubsub_mqtt_publish.c works, I don't have problem with this. I tried  to modify  this example to  make  a subscribe_mqtt,I added addReaderGroup,addDataSetReader ecc..but nothing, so I'm looking for an example in .c of subscribe_mqtt ( server version like publish_mqtt)

jackybek, 2021-07-19:
Can you drop me an email ?

davidem123, 2021-07-20:
> Can you drop me an email ?

davide.malena25@gmail.com

tony1411742631, 2021-12-23:
I'm looking for the same example. Do you have a such an example like subscriber_mqtt.c?

terkaa, 2022-02-01:
I would also like  to havetutorial_pubsub_mqtt_subscribe.c. I have already setup mqtt_publish so I can transfer joint position from Universal Robotics UR10 robot OPC UA plugin to Node-Red interface via MQTT and it is working. But now I would also like to send position commands from UI to robot :)

terkaa, 2022-02-01:
> Can you drop me an email ?

I would also like to havetutorial_pubsub_mqtt_subscribe.c. I have already setup mqtt_publish so I can transfer joint position from Universal Robotics UR10 robot OPC UA plugin to Node-Red interface via MQTT and it is working. But now I would also like to send position commands from UI to robot :)