open62541/open62541 #4431: Cannot build pubsub_interrupt_publish.c

santoshpanda1989, 2021-05-25:
## Description
I am trying to build the pubsub_interrupt_publish.c, but I am getting an error i.e. error: ‘UA_ServerConfig’ {aka ‘struct UA_ServerConfig’} has no member named ‘pubsubTransportLayers’
## Background Information / Reproduction Steps
I am building it in Ubuntu.

Used CMake options:

-->

```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_AMALGAMATION=OFF -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ETH_UADP=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_MALLOC_SINGLETON=ON -DUA_ENABLE_PUBSUB_BUFMALLOC=ON -DUA_ENABLE_IMMUTABLE_NODES=ON ..
```

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):master
 - [ x] Other OPC UA SDKs used (client or server): none
 - [ x] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ x] Critical issue No

NoelGraf, 2021-05-28:
The problem comes from a change in the API for simplifying the configuration of PubSub TransportLayers. Apparently, the example has not been adjusted yet.
You can replace the following:
```
config->pubsubTransportLayers = (UA_PubSubTransportLayer *)
        UA_malloc(sizeof(UA_PubSubTransportLayer));
config->pubsubTransportLayers[0] = UA_PubSubTransportLayerEthernet();
config->pubsubTransportLayersSize++;
```
with
```
UA_ServerConfig_addPubSubTransportLayer(config, UA_PubSubTransportLayerEthernet());
```

NoelGraf, 2021-06-04:
Has been fixed with #4445

closed 2021-06-04 by NoelGraf