open62541/open62541 #4434: No mechanism for multiple DataSetReader in PubSub ?

nqduy35, 2021-05-26:
## Background  
I implement a scenario including one server (subscriber) and two publishers. Each publisher sends only one data value to the subscriber. In the server-side, I create 1 Connection, 1 ReaderGroup, and 2 DataSetReader. The ReaderConfig of the two DataSetReader share publisherId, writerGroupId, but with two different datasetWriterId (e.g. 111 and 222). Running this scenario, the server have two nodes corresponding to each publisher. However, it can update only the data value from the one publisher, the second one cause `Error Write Value KF 0 ...`

## Problem found  
After checking the source code, I found that the functions UA_PubSubConnection_processNetworkMessage(...) and UA_DataSetReader_process(...) only support one UA_DataSetReader, so it is normal that my scenario gets error. But I'm not sure if I understand right or wrong, because the UA_ReaderGroup supports to add multiples DataSetReader. I mean UA_ReaderGroup has the attribute readersCount to count the number of associated DataSetReader. Do you have any suggestion to realise my scenario? Thank you in advance. 

## Checklist

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server):  uaexpert to check nodes in information model
 - [x] Operating system: linux

jpfr, 2021-06-04:
Fix and test via #4453.
Can you check this works for you?

nqduy35, 2021-06-07:
Thank you for considering my issue. I have checked with your fix, but the problem is still for my scenario: I can have data from the first publisher, but the second one is still with "Error Write Value KF 0 ...". However, as you have already tested it, so I will check my code once again. So I let this issue open for a while. 

Moreover, I found another issue: the example of the version 1.2 as in https://open62541.org/doc/current/server.html is no longer right with your new code. In detail, you have put pubsubTransportLayersSize and *pubsubTransportLayers into the 
UA_PubSubConfiguration, but the example is still for the old code.

o6-nbeier, 2026-10-20:
If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier