open62541/open62541 #4443: Valgrind memory leaks

EvgKlin, 2021-05-28:
## Description
I developed a OPCUA server with open62541 in Linux(Debian). I amalgamented files and i do not I received a segmentation fault. After some hours/Days memory increases. I start server with valgrind  and received the following output. 
In Issue https://github.com/open62541/open62541/issues/2343 was used `UA_ServerConfig_delete(config);` as solution, but i do not fimd this in new version
## Background Information / Reproduction Steps

==15182== HEAP SUMMARY:
==15182==     in use at exit: 827,191 bytes in 137,764 blocks
==15182==   total heap usage: 1,239,429 allocs, 1,101,665 frees, 18,143,840,819 bytes allocated
==15182==
==15182== 2 bytes in 1 blocks are definitely lost in loss record 1 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x118149: UA_String_fromChars (open62541.c:13717)
==15182==    by 0x110BC5: UA_LOCALIZEDTEXT_ALLOC (open62541.h:17243)
==15182==    by 0x3CC7FF: setDefaultConfig (open62541.c:191511)
==15182==    by 0x3CCE0B: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191800)
==15182==    by 0x110345: UA_ServerConfig_setMinimal (open62541.h:41772)
==15182==    by 0x1106E1: main (main.c:132)
==15182==
==15182== 9 bytes in 1 blocks are definitely lost in loss record 2 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x118149: UA_String_fromChars (open62541.c:13717)
==15182==    by 0x110603: main (main.c:112)
==15182==
==15182== 9 bytes in 1 blocks are definitely lost in loss record 3 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x118149: UA_String_fromChars (open62541.c:13717)
==15182==    by 0x3CC71B: setDefaultConfig (open62541.c:191496)
==15182==    by 0x3CCE0B: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191800)
==15182==    by 0x110345: UA_ServerConfig_setMinimal (open62541.h:41772)
==15182==    by 0x1106E1: main (main.c:132)
==15182==
==15182== 10 bytes in 2 blocks are indirectly lost in loss record 4 of 17
==15182==    at 0x4842260: calloc (vg_replace_malloc.c:752)
==15182==    by 0x11A30D: UA_Array_copy (open62541.c:14902)
==15182==    by 0x118267: String_copy (open62541.c:13760)
==15182==    by 0x119FE1: UA_copy (open62541.c:14765)
==15182==    by 0x110DF1: UA_String_copy (open62541.h:23097)
==15182==    by 0x3CADDB: UA_AccessControl_default (open62541.c:190442)
==15182==    by 0x3CCE89: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191820)
==15182==    by 0x110345: UA_ServerConfig_setMinimal (open62541.h:41772)
==15182==    by 0x1106E1: main (main.c:132)
==15182==
==15182== 13 bytes in 1 blocks are definitely lost in loss record 5 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x110561: fn_printConfigFile (main.c:91)
==15182==    by 0x110735: main (main.c:145)
==15182==
==15182== 17 bytes in 2 blocks are indirectly lost in loss record 6 of 17
==15182==    at 0x4842260: calloc (vg_replace_malloc.c:752)
==15182==    by 0x11A30D: UA_Array_copy (open62541.c:14902)
==15182==    by 0x118267: String_copy (open62541.c:13760)
==15182==    by 0x119FE1: UA_copy (open62541.c:14765)
==15182==    by 0x110DF1: UA_String_copy (open62541.h:23097)
==15182==    by 0x3CADF9: UA_AccessControl_default (open62541.c:190443)
==15182==    by 0x3CCE89: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191820)
==15182==    by 0x110345: UA_ServerConfig_setMinimal (open62541.h:41772)
==15182==    by 0x1106E1: main (main.c:132)
==15182==
==15182== 23 bytes in 1 blocks are definitely lost in loss record 7 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x118149: UA_String_fromChars (open62541.c:13717)
==15182==    by 0x3CC73D: setDefaultConfig (open62541.c:191497)
==15182==    by 0x3CCE0B: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191800)
==15182==    by 0x110345: UA_ServerConfig_setMinimal (open62541.h:41772)
==15182==    by 0x1106E1: main (main.c:132)
==15182==
==15182== 25 bytes in 1 blocks are indirectly lost in loss record 8 of 17
==15182==    at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182==    by 0x118149: UA_String_fromChars (open62541.c:13717)
==15182==    by 0x3CAEF1: UA_AccessControl_default (open62541.c:190471)
[... 82 lines cut ...]
==15182== LEAK SUMMARY:
==15182==    definitely lost: 826,682 bytes in 137,755 blocks
==15182==    indirectly lost: 157 bytes in 8 blocks
==15182==      possibly lost: 0 bytes in 0 blocks
==15182==    still reachable: 352 bytes in 1 blocks
==15182==         suppressed: 0 bytes in 0 blocks

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist
[ValgrindOutput.txt](https://github.com/open62541/open62541/files/6560173/ValgrindOutput.txt)

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue: I do not know

EvgKlin, 2021-05-28:
For example in function  UA_String_fromChars
```
UA_String_fromChars(const char *src) {
    UA_String s; s.length = 0; s.data = NULL;
    if(!src)
        return s;
    s.length = strlen(src);
    if(s.length > 0) {
        s.data = (u8*)UA_malloc(s.length);
        if(!s.data) {
            s.length = 0;
            return s;
        }
        memcpy(s.data, src, s.length);
    } else {
        s.data = (u8*)UA_EMPTY_ARRAY_SENTINEL;
    }

    return s;
}
```
Is missing UA_Free. Can i write it? or there is other fuctuion for this

mlang-de, 2021-05-28:
What does "is missing UA_Free" mean?
Normally UA_String_clear() shall be used to free an allocated string.

EvgKlin, 2021-05-28:
@mlang-de ok, i used ua_free(), now i am trying with UA_String_clear(). But i do not understand, why it commes memmory losses in open62541.c File. Im issue https://github.com/open62541/open62541/issues/2343 was meaned i should use UA_ServerConfig_delete(config), but in new version there is not such function

mlang-de, 2021-05-28:
May UA_ServerConfig_clean could solve your problem.

EvgKlin, 2021-05-28:
@mlang-de i tryed it, The same. I get maybe new or other errors, but it not helps

jpfr, 2021-05-28:
Valgrind indicates leaks from outside the library.

==15182== 826,476 bytes in 137,746 blocks are definitely lost in loss record 17 of 17
==15182== at 0x483F5C8: malloc (vg_replace_malloc.c:299)
==15182== by 0x10DE2B: myGetUserAccessLevel (AccessControl.c:28)

EvgKlin, 2021-05-30:
c28. is char *chAdminLogin =(char *)UA_malloc(sizeof(char) * logins[ADMINUSER].username.length + 1);
//..//
UA_free(chAdminLogin); // in the same Function

EvgKlin, 2021-05-31:
ok, i find some errors, but i get always mistake in 
```

==30245==    by 0x3CCB59: UA_ServerConfig_setMinimalCustomBuffer (open62541.c:191820)
==30245==    by 0x110141: UA_ServerConfig_setMinimal (open62541.h:41772)
==30245==    by 0x11015F: UA_ServerConfig_setDefault (open62541.h:41796)
==30245==    by 0x110477: main (main.c:134)
```
this error lead to other errors like malloc in open62541.c

EvgKlin, 2021-05-31:
@jpfr i am not sure, where can be the errors. And I do not know, how can i fix or how can i find the errors.  It seems that the memory leaks are from methods that I call from my main method. I write some my source code. Maybe you habe some suggestion
Or there some other ways to find the errors

[Valgring_ohne_Client1.txt](https://github.com/open62541/open62541/files/6571105/Valgring_ohne_Client1.txt)

[Valgring_with_Client.txt](https://github.com/open62541/open62541/files/6571107/Valgring_with_Client.txt)
[example_server.txt](https://github.com/open62541/open62541/files/6571129/example_server.txt)

closed 2021-06-02 by EvgKlin

mlang-de, 2021-06-02:
What was the problem in your case? Maybe it could be interessting for other users in future too.

EvgKlin, 2021-06-02:
you are right, errors are stupid because of dirty code and other.
1) my errors with memory corruption
2) do not clear like config like accessControl or applicationuri/buildInfo
3)use specific function for strings like UA_String_equal, UA_String_copy, UA_String_clear
4) some other, at the moment i am not finding it