open62541/open62541 #4461: Call Method to change Variable

Psoke, 2021-06-08:
## Description
Hey there,
I have a little question regarding the use of methods. I am running an open62541 server on a laptop in my network. Now I want to connect to that server via client and call a method that changes a variable on that server.
Lets say I have a variable "age" with the value 10 and a method "travelintimesoyoucandrinklegally" that changes the value to 21.

Is that possible and could you point me to some sort of documentation or tutorials?
With best regards
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

EvgKlin, 2021-06-09:
https://github.com/open62541/open62541/blob/master/examples/tutorial_server_method.c

to call methode from client https://github.com/open62541/open62541/blob/4e60f4e5e8543cdbf3e9e0c96a20e8ec8681401f/examples/client.c#L173

Psoke, 2021-06-14:
Ok thank you so far.

I have another question: 

https://github.com/open62541/open62541/blob/51686828126d805fe72bc6021634b9fea382df37/examples/tutorial_server_method.c#L41-L88

If i would like to model my own information model using a GUI like freeopcuamodeler, is it possible to connect a pre-established method with a callback?
So basically I add my Method via freeopcuamodeler what makes addHellWorldMethod() unnecessary. But how do I connect the input and output argument established in freeopcuamodeler to the callback? What if I want to use my own namespace?

My main problem here is that I do not understand how the addHellWorldMethod() (which simply adds a method to the server) is connected and or calls the function helloWorldMethodCallback(). I see in line 86 that the adress of the function is referenced via &helloWorldMethodCallback but what does that do? Does it simply connect the method to the callback or just to the adress where the input and output are stored or how does it work?

My goal: I just want to code callbacks in my server main.c which I can connect to methods with their in- and outputs that are pre-established by a GUI. 

Is there maybe a more precise documentation on all the functions available in open62541 where the passed parameter and behaviour is explained?

jpfr, 2021-06-16:
> Is there maybe a more precise documentation on all the functions available in open62541 where the passed parameter and behaviour is explained?

This is documented in the server.h which is also translated into the docs.

So you want to use an OPC UA modeller to also define the "backend callbacks".
Right now, the standard XML format that is created from the modeller does not contain such information.

You could internally have a table that maps from method names to callback pointers.
And then loop over that at the beginning to add the callbacks.
But you'd have to program that. We don't have such a helper in the code base right now.

andreasebner, 2025-07-30:
Question seems to be answered. Feel free to reopen if needed

closed 2025-07-30 by andreasebner