open62541/open62541 #4462: Encoding of empty array of Extension object

Stasik0, 2021-06-08:
- [x ] open62541 Version (release number or git tag): 1.2

Dear community,

just one generic question. In the method callback I need to return an empty array of extension objects.

I use the following code
`UA_Variant_setArray(&(output[0]), NULL, 0, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT]); `

however wireshark shows "-1" as transmitted array length.

[image]

In case 0 length is invalid, is it something I missed in the spec? In case it is valid - how do I set it from the API?

jpfr, 2021-06-12:
We use size_t, an unsigned integer to track array sizes internally.

- length 0, NULL pointer -> -1 array
- length 0, UA_EMPTY_ARRAY_SENTINEL pointer -> 0 array

You can use UA_Array_new to generate an array of length 0.
That automatically gives a UA_EMPTY_ARRAY_SENTINEL pointer.

Stasik0, 2021-06-12:
Thanks! But is -1 a valid value „on wire“?

jpfr, 2021-06-16:
Yes. On the wire.
But we use an unsigned size_t for array sizes internally.

Reopen if necessary.

closed 2021-06-16 by jpfr