open62541/open62541 #4468: LibreSSL support

ccvca, 2021-06-11:
## Description
As building openSSL on windows (mingw-w64 without MSYS2) is a pain, I checked out LibreSSL which CMake-based build is quite easy. But compiling open62541 was not possible out of the box.

**Does anyone has success building open62541 with LibreSSL?
Is a LibreSSL support desired?**

I've applied several hacks to get it working, but it was quite complicated: https://github.com/ccvca/open62541/tree/libreSSL

- OPENSSL_VERSION_NUMBER Checks are broken (https://github.com/libressl-portable/portable/issues/573)

> That said, I'm not sure why someone would write code that uses `(OPENSSL_VERSION_NUMBER >= 0x10100000L) && (OPENSSL_VERSION_NUMBER < 0x20000000L)` - that seems like a rather strange version check, even by OpenSSL standards.

 -  -Werr is always triggered by  `#warning overriding WinCrypt defines`

Hack-Solution:
```c
/* Hack: Prevent Wincrypt-Includes */
#define __WINCRYPT_H__
# include <iphlpapi.h>
#undef __WINCRYPT_H__
```

## Background Information / Reproduction Steps

Used CMake options:

- "UA_ENABLE_ENCRYPTION_OPENSSL": "ON"
-  "UA_ENABLE_ENCRYPTION": "ON"
-  "UA_BUILD_EXAMPLES": "ON"

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 4e60f4e5e8543cdbf3e9e0c96a20e8ec8681401f
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win 10, mingw-w64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

undroidwish, 2021-06-11:
You might want to try out my patch which allows me to build with libressl:

https://www.androwish.org/home/file?name=jni/topcua/open62541/chw.patch&ci=tip

ccvca, 2021-06-11:
It looks like you've disabled some security checks and avoid the "Werr" Problem by using the amalganated build with custom settings.

undroidwish, 2021-06-11:
Maybe, surely won't I remember. Use that stuff as a starting point and give us hints how to do it proper.

PR #4529 (2021-07-14): [Review]: feat(plugin): Add LibreSSL support
As building openssl with mingw on windows can be quite difficult, this PR introduces LibreSSL support reusing the openssl implementation.

To be compliant with `-Werr`, a little hack is required in `win32/ua_architecture.h` as WinCrypt must not be included together with LibreSSL:
https://github.com/open62541/open62541/blob/daf3b765e6dd0aafe373cfca7a016576b696cba0/arch/win32/ua_architecture.h#L117-L130

CI: Change to Matrix-Build on github to reduce time and install minimal dependencies (installing openssl may conflict with libressl)

Resolve: #4468
closed 2021-07-14 by jpfr