open62541/open62541 #4470: UA_parseEndpointUrl() succeeds for (?)invalid(?) URL

jfischer1, 2021-06-11:
## Description
During testing I tried to connect via open6541 client to a server using URL opc.tcp:///127.0.0.1:4840. To my knowledge, this URL is not valid due to 3 slashes after protocol specifier. However, connecting to server worked. Debugging lead me to UA_parseEndpointUrl(). Here are the relevant code lines (similar to code found in network_tcp.c):
```
    UA_String hostnameString = UA_STRING_NULL;
    UA_String pathString = UA_STRING_NULL;
    UA_UInt16 port = 0;
    UA_String endpointUrl = UA_STRING("opc.tcp:///127.0.0.1:4840");
    UA_StatusCode status = UA_parseEndpointUrl(&endpointUrl, &hostnameString, &port, &pathString);
```
UA_parseEndpointUrl() returned UA_STATUSCODE_GOOD. Is this the correct/intended?

## Background Information / Reproduction Steps
Using open62541 version 1.2.0. Platform Windows, Visual Studio 2015.

ccvca, 2021-06-15:
Just by loooking in the code, I think, that `/127.0.0.1:4840` is seen as the pathString-Argument:
https://github.com/open62541/open62541/blob/a2a63f87dbd4a8e248c176cb053f825e09694796/src/ua_util.c#L76

Have you checked the returned arguments?

jfischer1, 2021-06-15:
The code snippet yields the following values:

- status = 0
- hostnameString
-- length 0
-- data = "/127.0.0.1:4840"
- port = 0
- pathString
-- length = 14
-- data = 127.0.0.1:4840

ccvca, 2021-06-15:
Thank you, this is definitely a bug.