open62541/open62541 #4475: Frequent connection breaks due to sequence number mismatches

markekk, 2021-06-17:
## Description
OPC UA client connection to server breaks frequently due to following error "Processing the message failed with error BadSecurityChecksFailed".  Connection security policy is SecurityPolicy#None. 

It seems that this error comes from: ua_securechannel.c/processSequenceNumberSym.

As far as I know, this problem does not occur with open62541 v1.0

## Background Information / Reproduction Steps
About problem occurring on "production" environment:
- Both OPC UA client and server are running in same Ubuntu 16.04
- Client monitors all  the variables that are available on server (maybe 300)
- Client sends "burst of writes" to maybe 50 variables. Delay between each write is about couple of milliseconds. Meaning that 50 variables are written in 50-100ms. These "write bursts" are not done constantly. It seems that BadSecurityChecksFailed error occurs during "burst".
=> problem occurs daily

About problem occurring on "development" environment:
- Both OPC UA client and server are running in same Ubuntu 16.04 VirtualBox environment. 
- There is only couple of variables on server interface. 
- Client monitors all  the variables that are available on server 
- Client sends constantly writes to one variable using about 2ms interval
=> it does not take more than 60s seconds to reproduce problem

## Log entries
Client:
17.06.21 09:29:47.379>[open62541 W] "Connection 27 | SecureChannel 2 | Receiving the response failed with StatusCode BadSecurityChecksFailed"
17.06.21 09:29:47.379>[open62541 W] "Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed"

Server:
17.06.21 09:29:47.379>[open62541 I] "Connection 170 | SecureChannel 2 | CloseSecureChannel"
17.06.21 09:29:47.380>[open62541 I] "Connection 171 | New connection over TCP from 127.0.0.1"
17.06.21 09:29:47.380>[open62541 I] "Connection 170 | Closed"
17.06.21 09:29:47.381>[open62541 I] "Connection 171 | SecureChannel 3 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s"
17.06.21 09:29:47.382>[open62541 I] "SecureChannel 3 | Session ns=1;g=98eec1b6-e19d-b494-8f28-92e9714764e6 | ActivateSession: Session attached to new channel"
17.06.21 09:29:47.382>[open62541 I] "SecureChannel 3 | Session ns=1;g=98eec1b6-e19d-b494-8f28-92e9714764e6 | ActivateSession: Session activated"

Added custom logging to ua_securechannel.c and noticed that there could be problem with sequence number:
[image]

## Used CMake options:
cmake ../open62541 -DUA_ENABLE_AMALGAMATION=Off -DCMAKE_BUILD_TYPE=Release -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=OFF -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_ENCRYPTION=On -DBUILD_SHARED_LIBS=On -DCMAKE_INSTALL_PREFIX:PATH=../install

## Checklist
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.2.2, 1.1.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Xubuntu 16.04.7
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-07-02:
Looking into that.
What I couldn't gather from the logs is: Is the sequence number mismatch reported on the server-side or the client-side?

markekk, 2021-07-02:
Hi @jpfr, I originally forgot to document that on which side the mismatch was detected/reported. I reproduced the problem and it seems mismatch is detected/reported on server side:

Server log:
11:58:19.450Z Debug>[open62541 T] "Connection 11 | Activity on the socket"
11:58:19.450Z Debug>[open62541 T] "Connection 11 | Received a packet."
11:58:19.450Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1626 of type WriteResponse"
11:58:19.450Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1627 of type WriteResponse"
11:58:19.450Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.450Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1628 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1629 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1630 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1631 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1632 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1633 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Write attribute 13 of Node ns=1;s=TEST.VARIABLE"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Sending response for RequestId 1634 of type WriteResponse"
11:58:19.451Z Debug>[open62541 T] "Connection 11 | SecureChannel 2 | Process a MSG"
11:58:19.451Z Debug>[open62541 D] "SecureChannel 2 | Session ns=1;g=fc3aecac-508d-251d-0330-33332aa29ac9 | Processing WriteRequest"
[... cut ...]
11:58:19.500Z Critical>[open62541 E] "Connection 16 | SecureChannel 2 | Received an ERR response with StatusCode BadSecurityChecksFailed and the following reason: "
11:58:19.500Z Warning>[open62541 W] "Connection 16 | SecureChannel 2 | Receiving the response failed with StatusCode Good"
11:58:19.500Z Warning>[open62541 W] "Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed"
11:58:19.500Z Info>[open62541 I] "Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: BadSecurityChecksFailed"

markekk, 2021-07-19:
hi @jpfr, I was able to reproduce similar problems using following new test client application with old server application. 

Here is the code of client application:
[image]

Communication problems will occur quite fast after starting client. Usually one of these problems will occur:
- Connection will break due to BadSecurityChecksFailed error
- Communication gets stuck 

hmm... is it OK to call UA_Client_writeValueAttribute_async() from separate thread while UA_Client_run_iterate() executed?

markekk, 2021-07-30:
I modified my application code so that only one thread is used for open62541 communication. It seems that probability of problems have decreased significantly. 

It seems that there is still connection breaks even server and client are running on same machines. It can take about two hours until first connection break occurs. I'll need to do more testing...

tilinna, 2021-08-24:
I'm running a similar open62541 v1.2.2-based single-thread -client against a python-opcua test server (Basic256Sha256 with userpass) and get the same BadSecurityChecksFailed -error every now and then. The client is subscribed only to the server time ("i=2258") node.

I noticed that if I decrease the secureChannelLifeTime in the UA_ClientConfig to 1000ms, the problem appears much more often. After the failure, open62541 renews the session by itself, but loses active subscriptions (unsure if that is by design).

The Python server reports a sequence problem:

opcua.ua.uaerrors._base.UaError: Received chunk: MessageChunk(Header(type:b'CLO', chunk_type:b'F', body_size:84, channel:75), SequenceHeader(SequenceNumber:22, RequestId:25 ), SymmetricAlgorithmHeader(TokenId:15 ), 35 bytes) with wrong sequence expecting: 20, received: 22, spec says to close connection

Is it possible that a renewal of the secure channel is somehow interfered by a server sending new node values in the middle of the process?

jvallesf, 2022-11-11:
In case this helps somebody, I have recently refactored the implementation of the OPC Client to single-threaded to make it more robust to this kind of errors. I was still having frequent errors with returned status BadSecurityChecksFailed and BadConnectionLost when writing to the server very fast... until I finally found I still had a call to UA_Variant_setScalarCopy from a different thread. I corrected it and now works perfectly, without errors.

closed 2025-07-30 by andreasebner