open62541/open62541 #4482: Can not change variable with methode

EvgKlin, 2021-06-21:
## Description
I create Object-Node with Variable. ParentNodeId of Variable is outNewNodeId of Project. After that i added methode with ParentID of Object. In my callback function i want to change variable, but i get error. I do not undertan the error
## Background Information / Reproduction Steps
```

UA_NodeId DIOutID00;
  UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Digitale Output00");
    UA_Server_addObjectNode(
        server, UA_NODEID_NULL, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
        UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
        UA_QUALIFIEDNAME(2, "Digitale Output00"),
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), oAttr, NULL, &DIOutID00);

UA_StatusCode retCode = UA_Server_addVariableNode(
        server, myIntegerNodeId, DIOutID00, parentReferenceNodeId, myIntegerName,
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

void
AddMethodDI00(UA_Server *server) {
    UA_Argument inputArgument;
    UA_Argument_init(&inputArgument);
    inputArgument.description = UA_LOCALIZEDTEXT("en-US", "Unsigned Integer");
    inputArgument.name = UA_STRING("Input");
    inputArgument.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    inputArgument.valueRank = UA_VALUERANK_SCALAR;

    UA_Argument outputArgument;
    UA_Argument_init(&outputArgument);
    outputArgument.description = UA_LOCALIZEDTEXT("en-US", "Unsigned Integer");
    outputArgument.name = UA_STRING("Output");
    outputArgument.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    outputArgument.valueRank = UA_VALUERANK_SCALAR;

    UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
    helloAttr.description = UA_LOCALIZEDTEXT("en-US", "Methode for digital Output");
    helloAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Methode for digital Output");
    helloAttr.executable = true;
    helloAttr.userExecutable = true;
    UA_Server_addMethodNode(server, UA_NODEID_STRING(2, "DiOutput00"), DIOutID00,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                            UA_QUALIFIEDNAME(2, "Methode for digital Output"), helloAttr,
                            &MethodCallback00, 1, &inputArgument, 1, &outputArgument,
                            NULL, NULL);
}
static UA_StatusCode
MethodCallback01(UA_Server *server, const UA_NodeId *sessionId, void *sessionHandle,
                 const UA_NodeId *methodId, void *methodContext,
                 const UA_NodeId *objectId, void *objectContext, size_t inputSize,
                 const UA_Variant *input, size_t outputSize, UA_Variant *output) {
    UA_Int32 *inputStr = (UA_Int32 *)input->data;

    /*UA_String tmp = UA_STRING_ALLOC("Hello ");*/

    UA_StatusCode retval =
        UA_Variant_setScalarCopy(output, inputStr, &UA_TYPES[UA_TYPES_INT32]);

    UA_NodeId currentNode = UA_NODEID_STRING(2, "DigitaleOutput15");
    retval=UA_Server_writeValue(server, currentNode, *output);
    return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
MethodCallback00(UA_Server *server, const UA_NodeId *sessionId, void *sessionHandle,
                 const UA_NodeId *methodId, void *methodContext,
                 const UA_NodeId *objectId, void *objectContext, size_t inputSize,
                 const UA_Variant *input, size_t outputSize, UA_Variant *output) {
    UA_Int32 *inputStr = (UA_Int32 *)input->data;
    UA_Int32 inputs = 16;

    //UA_Variant myVar;
    //UA_Variant_init(&myVar);
    //UA_Variant_setScalar(&myVar, &inputs, &UA_TYPES[UA_TYPES_INT32]);
    //retval= UA_Server_writeValue(server, currentNode, myVar);
    /* Write a different integer value */
    UA_Int32 myInteger = 43;
    UA_Variant myVar;
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    retval=UA_Server_writeValue(server, currentNode, myVar);
    /*UA_Variant_setScalarCopy(output, &tmp, &UA_TYPES[UA_TYPES_STRING]);
    UA_String_clear(&tmp);*/

    return UA_STATUSCODE_GOOD;
}
```
Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

embbo, 2021-06-22:
@EvgKlin check this example ..
https://github.com/open62541/open62541/blob/master/examples/tutorial_server_method.c

EvgKlin, 2021-06-22:
i do it with this example =)

embbo, 2021-06-22:
@EvgKlin what error you get ? can you please elobrate ?