open62541/open62541 #4489: undefined reference to

embbo, 2021-06-24:
## Description
I just complied the latest files .. cmake and make .. and obtain open62541.h/62541.c ...but i got these errors .. I am using eclipse ..but old files works fine ..

/../Src/open62541.c:70232: undefined reference to `UA_SecurityPolicy_Basic128Rsa15'
/../Src/open62541.c:70266: undefined reference to `UA_SecurityPolicy_Basic256'
/../Src/open62541.c:70300: undefined reference to `UA_SecurityPolicy_Basic256Sha256'
/../Src/open62541.c:70334: undefined reference to `UA_SecurityPolicy_Aes128Sha256RsaOaep'
/../Src/open62541.c:70417: undefined reference to `UA_CertificateVerification_Trustlist'

and these are declared in open62541.h

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-07-19:
You need to Enable the CCmake options relating to encryption.

embbo, 2021-07-21:
@jackybek  i did thats why it is  calling that funtion...

jackybek, 2021-07-21:
Can you list the ccmake options that are enabled ? 
Did you turn on almagamation option ?
Did you turn on built shared lib option ?
In my case, if I use almagamation and don’t enable shared lib, I don’t encounter missing references, but the moment I turn off shared lib, I get funny issues similar to yours.

embbo, 2021-08-09:
@jackybek yes i did , actually I make other old version it works fine ... but in this we both have same issue ... i don't know why

embbo, 2021-08-09:
@jackybek Yes without encryption , it works fines but with library ...it gives that issues

closed 2025-07-30 by andreasebner