open62541/open62541 #4492: How can I change the IP-adress during runtime?

Max15421, 2021-06-28:
## Description
Hi, I want to change the IP adress of my OPC UA Server during runtime. I want an OPC UA client to set a value at a node which should change the ip adress of the server. Thats how I want to configure the IP adress. Is this possible or is there another way to configure the ip adress using opc? 

Thanks and Cheers
Max
## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server):  UAExpert
 - [ ] Operating system: FreeRTOS+ lwIP
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2021-07-19:
Changing the IP address of a server can be done through the following code segment ::
HTTPS://stackoverflow.com/questions/6652384/how-to-set-the-ip-address-from-c-in-linux 

you can implement the above as a method call but I am not sure what will happen to the connected clients when the server Ip changes, especially how lds will react to this change.

andreasebner, 2025-07-30:
There's been no further comment, so I expect this issue to be resolved.