open62541/open62541 #4493: Support multiple DSR during freeze configuration

mada-github, 2021-06-28:
## Description
Support multiple DSR in freezeReaderGroupConfiguration (same for writer groups)

## Background Information / Reproduction Steps
see TODO in source code
```C
UA_Server_freezeReaderGroupConfiguration()
[…]
// TODO: Clarify on the freeze functionality in multiple DSR, multiple networkMessage conf in a RG
[…]
if(dsrCount > 1) {
  UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_SERVER,
                 "Mutiple DSR in a readerGroup not supported in RT fixed size configuration");
  return UA_STATUSCODE_BADNOTIMPLEMENTED;
}
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-01-30:
@andreasebner please have a look

jpfr, 2024-02-15:
@gauravks please have a look