open62541/open62541 #4494: UA_Variant_print or equivalent: does it exists / is there interest to propose it?

pnikiel, 2021-06-29:
Hi,

I was looking for an unified way of printing data contained in UA_Variant; those of you who know UASDK know that they for instance have UaVariant::toString which is super handy for diagnostics, debugging, etc, etc.

But I haven't found anything like this.

Does it exist, if not is there interest in a contribution and what would be the constraints?

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-06-30:
Did you see this?
```c
UA_StatusCode
UA_print(const void *p, const UA_DataType *type, UA_String *output);
```

jpfr, 2021-07-06:
Just reopen if required.
Contributions are of course welcome!

closed 2021-07-06 by jpfr