open62541/open62541 #4497: Error when building package that uses open62541

joe28965, 2021-06-30:
## Description
I have written my OPC UA client (opcua_bridge) using open62541. However, when I try and build it (using colcon build) I get the following error:

```
/usr/bin/ld: cannot find -lmbedtls
/usr/bin/ld: cannot find -lmbedx509
/usr/bin/ld: cannot find -lmbedcrypto
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/opcua_bridge.dir/build.make:152: opcua_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/opcua_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

## Background Information / Reproduction Steps
Operating system: Ubuntu 20.04
Installed using the [debian](https://open62541.org/doc/current/installing.html#debian)

## Fix

I already found how to fix it, I need to run the following:

`sudo apt install libmbedtls-dev`

However, it's not mentioned on the installation page. I also feel like it should be something that's installed automatically when you install open62541.

If you have any other questions, please let me know. I'll be happy to provide more information

jpfr, 2021-06-30:
Which version of the library did you install?
Where did you download it?

joe28965, 2021-06-30:
I installed it using the instructions for the debian installation on the [website](https://open62541.org/doc/current/installing.html#debian):

```
sudo add-apt-repository ppa:open62541-team/ppa
sudo apt-get update
sudo apt-get install libopen62541-1-dev
```

Which, if I'm not mistaken, installed version 1.1.5. (at least, when I run `sudo apt-get install libopen62541-1-dev` again, I get `libopen62541-1-dev is already the newest version (20210128T143534~1.1.5~1~ge1650d8a~ubuntu20.04.1).`

jackybek, 2021-07-19:
Julius - actually the command ‘sudo apt-get install … limbedtls-dev ‘  … can be found in www.kalycito.com/running-opcua-on-raspberry-pi/ as part of the installation steps.  You may want to update the online documents in open62541.org

closed 2022-04-08 by joe28965