open62541/open62541 #4498: typeIndex in code

nokem61, 2021-06-30:
## Description

ua_services_attribute.c line 1728

```
const UA_DataType *historyDataType = &UA_TYPES[UA_TYPES_HISTORYDATA];
    UA_HistoryDatabase_readFunc readHistory = NULL;
    switch(request->historyReadDetails.content.decoded.type->typeIndex)
```

error C2039: 'typeIndex': is not a member of 'UA_DataType'

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2021-06-30:
Jup. That's a bug.
It seems we don't have the historizing enabled in the CI.
Let's correct that.

nugatritter, 2021-06-30:
Will UA_DataType::typeIndex and UA_DataTypeMember::memberTypeIndex come back?
They are essential for handling custom data types in client applications!
Please do not violate any important interface definitions!

jpfr, 2021-07-02:
The fix comes via #4501.

@nugatritter We now have a pointer to the datatype of every member.
That is more powerful and more flexible than the indexing we had before.

Sorry that you have to make local adjustments. We had no choice but to change the definition.
The typeIndex approach was not powerful enough to let datatypes from different companions specs (.bsd files) reference each other as members. We need that to support the latest companion specs.

I am very confident that every previous use case can be made to work with this structure as well.
Feel free to ask specific questions.

closed 2021-07-02 by nokem61

nugatritter, 2021-07-02:
Hello Julius,

thanks for your quick reply. I'm a member of the opcua - EPICS Device 
Support for OPC UA <https://github.com/paulscherrerinstitute/opcua> 
team. My task is to implement custom defined data types for a client 
application. And here comes my question: Where can I find out more about 
planned changes such as changes in the UA_DataType(Member) structures?

Best regards