open62541/open62541 #4499: client_connect does not connect with user and password to UA-SERVER

POWER-KI, 2021-06-30:
## Description

Having problems with my app to connect with User and Psw to a server I have tried with the example "client_connect" and tested against UA-SERVER Expert Demo:

`P:\XU\PWK\WIN\OPC-UA\BUILD\bin\examples>client_connect -username john -password master -securityMode 2 -securityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 opc.tcp://192.168.2.4:48010

[2021-06-30 13:23:11.943 (UTC+0200)] warn/userland AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-06-30 13:23:11.959 (UTC+0200)] info/userland ID john - Psw master - Policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2021-06-30 13:23:11.959 (UTC+0200)] info/channel Connection 272 | SecureChannel 3393113543 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-06-30 13:23:11.959 (UTC+0200)] info/client Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 0: security mode doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 1: security policy not available
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 2: security mode doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 3: security policy doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 4: security mode doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 5: security policy doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 6: security mode doesn't match
[2021-06-30 13:23:11.975 (UTC+0200)] error/client No suitable endpoint found
[2021-06-30 13:23:11.975 (UTC+0200)] info/client Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
`
Note that UA-Client works. I have tried many combination of secure but always with no results.
My openLib use mbedTSL as crypto provider.

## Background Information / Reproduction Steps

`
Note that UA-Client works. I have tried many combination of secure but always with no results.
My openLib use mbedTSL as crypto provider.

Thank you for any help !

CAP

jpfr, 2021-07-06:
Please check with UA Expert or similar which endpoints / SecurityPolicies are available.

liutt1312, 2021-08-12:
> ## 描述
> 我的应用程序在将 User 和 Psw 连接到服务器时遇到问题，我尝试使用示例“client_connect”并针对 UA-SERVER Expert Demo 进行了测试：
> 
> `P:\XU\PWK\WIN\OPC-UA\BUILD\bin\examples>client_connect -username john -password master -securityMode 2 -securityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 opc.tcp:/ /192.168.2.4:48010
> 
> [2021-06-30 13:23:11.943 (UTC+0200)] 警告/用户域 AcceptAll 证书验证。任何远程证书都将被接受。
> [2021-06-30 13:23:11.959 (UTC+0200)] info/userland ID john - Psw master - Policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
> [2021-06-30 13:23: 11.959 (UTC+0200)] 信息/频道连接 272 | 安全通道 3393113543 | SecureChannel 以 SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None开放，修改后的生命周期为 600.00s
> [2021-06-30 13:23:11.959 (UTC+0200)] info/client Client Status: ChannelState: Open , SessionState: Closed, ConnectStatus: Good
> [2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 0: security mode does not match
> [2021-06-30 13:23:11.975 (UTC+0200)] info/client
> Rejecting endpoint 1: security policy not available [2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 2：安全模式不匹配
> [2021-06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 3：安全策略不匹配
> [2021-06-30 13:23:11.975 ( UTC+0200)] info/client 拒绝端点 4：安全模式不匹配
> [2021-06-30 13:23:11.975 (UTC+0200)] info/client 拒绝端点 5：安全策略不匹配
> [2021 -06-30 13:23:11.975 (UTC+0200)] info/client Rejecting endpoint 6: security mode does not match
> [2021-06-30 13:23:11.975 (UTC+0200)] 错误/客户端不合适找到端点
> [2021-06-30 13:23:11.975 (UTC+0200)] info/client Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
> `
> 
> 请注意，UA-Client 有效。我尝试了许多安全的组合，但总是没有结果。
> 我的 openLib 使用 mbedTSL 作为加密提供程序。
> 
> ## 背景资料/复制步骤
> `
> open62541 ver 1.2
> 注意UA-Client 可以工作。我尝试了许多安全的组合，但总是没有结果。
> 我的 openLib 使用 mbedTSL 作为加密提供程序。
> 
> 感谢您的任何帮助 ！
> 
> 帽

need cert and key

smitha-cgi, 2021-08-30:
Make sure you have included Basic128 and Basic256 in your supported security policies as I have found that authenticated connections won't work without them:

```
if (config->securityPolicies EQ NULL)
{
    status = UA_ClientConfig_setDefaultEncryption(config, certificate, privateKey, trustList, trustListSize, revocationList, revocationListSize);
    if (status EQ UA_STATUSCODE_GOOD)
    {						
        //UA_ClientConfig_setDefaultEncryption resets config back to default so reapply settings
	config->logger.log = UA_Log;
        config->timeout = m_ScanTimeout * 1000;

	if ((m_Username.length() GT 0) AND (m_Password.length() GT 0))
	{
	    config->securityPoliciesSize = 4;	//None, 256Sha256, Basic128 and Basic256 (required for authentication)
	}
	else
	{
	    config->securityPoliciesSize = 2;	//None and 256Sha256
	}

	config->securityPolicies = new UA_SecurityPolicy[config->securityPoliciesSize];
	UA_SecurityPolicy_None(&config->securityPolicies[0], UA_BYTESTRING_NULL, &config->logger);

	//Configure a secure connection with Basic256Sha256
	config->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
	status = UA_SecurityPolicy_Basic256Sha256(&config->securityPolicies[1], certificate, privateKey, &config->logger);

	if ((m_Username.length() GT 0) AND (m_Password.length() GT 0))
	{
	    UA_SecurityPolicy_Basic128Rsa15(&config->securityPolicies[2], certificate, privateKey, &config->logger);
	    UA_SecurityPolicy_Basic256(&config->securityPolicies[3], certificate, privateKey, &config->logger);
	 }
    }
}	
```

jpfr, 2021-09-23:
Closing due to inactivity.

closed 2021-09-23 by jpfr