open62541/open62541 #4502: Subscription stops working if connection to server is re-established after an interruption (i.e. due to network)

smitha-cgi, 2021-07-02:
## Description
When a client creates a subscription to get changes from a server, once the connection to the server is interrupted and re-established, the subscription no longer works.

## Background Information / Reproduction Steps

1. Set up an OPC UA server to increment a value once per second (I used the MyLevel object in Prosys OPC UA Simulation Server). Run server on a different machine to the client.
2. Run open62541 client and create a subscription for the changing object using default settings. Print out a message when data is received.
3. Disable the network connection between the two hosts so that the server is still ticking along generating changes. 
4. Re-establish the network connection between the two hosts. No data is returned and a message is printed out every 10 seconds: "Inactivity for Subscription xx."

Log showing the issue:
```
[2021-07-02 17:54:12.273 (UTC+1000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2021-07-02 17:54:12.341 (UTC+1000)] info/channel       Connection 144 | SecureChannel 47 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-07-02 17:54:12.350 (UTC+1000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2021-07-02 17:54:12.365 (UTC+1000)] info/client        Rejecting endpoint 0: security policy not available
[2021-07-02 17:54:12.366 (UTC+1000)] info/client        Rejecting endpoint 1: security policy not available
[2021-07-02 17:54:12.369 (UTC+1000)] info/client        Rejecting UserTokenPolicy 0 in endpoint 2: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2021-07-02 17:54:12.370 (UTC+1000)] info/client        Selected Endpoint opc.tcp://x.x.x.x:53530/OPCUA/SimulationServer with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-07-02 17:54:12.374 (UTC+1000)] info/client        Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2021-07-02 17:54:12.506 (UTC+1000)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2021-07-02 17:54:12.533 (UTC+1000)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
Create subscription succeeded, id 31
Monitoring 'MyLevel', id 1

MyLevel has changed - new value 43.00 - 2/7/2021 7:54:12
MyLevel has changed - new value 44.00 - 2/7/2021 7:54:13
MyLevel has changed - new value 45.00 - 2/7/2021 7:54:14
MyLevel has changed - new value 46.00 - 2/7/2021 7:54:15
MyLevel has changed - new value 47.00 - 2/7/2021 7:54:16
MyLevel has changed - new value 48.00 - 2/7/2021 7:54:17
MyLevel has changed - new value 49.00 - 2/7/2021 7:54:18
MyLevel has changed - new value 50.00 - 2/7/2021 7:54:19
MyLevel has changed - new value 51.00 - 2/7/2021 7:54:20
MyLevel has changed - new value 52.00 - 2/7/2021 7:54:21
MyLevel has changed - new value 53.00 - 2/7/2021 7:54:22
MyLevel has changed - new value 54.00 - 2/7/2021 7:54:23
MyLevel has changed - new value 55.00 - 2/7/2021 7:54:24
MyLevel has changed - new value 56.00 - 2/7/2021 7:54:25
MyLevel has changed - new value 57.00 - 2/7/2021 7:54:26
MyLevel has changed - new value 58.00 - 2/7/2021 7:54:27
[2021-07-02 17:54:37.913 (UTC+1000)] warn/channel       Connection 144 | SecureChannel 47 | Receiving the response failed with StatusCode BadConnectionClosed
[2021-07-02 17:54:37.916 (UTC+1000)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2021-07-02 17:54:37.921 (UTC+1000)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
[2021-07-02 17:54:42.923 (UTC+1000)] warn/network       Connection to opc.tcp://x.x.x.x:53530/OPCUA/SimulationServer failed with error:Unknown error
[2021-07-02 17:54:42.928 (UTC+1000)] info/client        Client Status: ChannelState: Fresh, SessionState: Created, ConnectStatus: BadDisconnect
[2021-07-02 17:54:47.938 (UTC+1000)] warn/network       Connection to opc.tcp://x.x.x.x:53530/OPCUA/SimulationServer failed with error:Unknown error
[2021-07-02 17:54:47.943 (UTC+1000)] info/client        Client Status: ChannelState: Fresh, SessionState: Created, ConnectStatus: BadDisconnect
[2021-07-02 17:54:47.961 (UTC+1000)] info/channel       Connection 144 | SecureChannel 48 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2021-07-02 17:54:47.966 (UTC+1000)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2021-07-02 17:54:47.988 (UTC+1000)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2021-07-02 17:54:48.991 (UTC+1000)] error/client       Inactivity for Subscription 31.
[2021-07-02 17:54:58.995 (UTC+1000)] error/client       Inactivity for Subscription 31.
```

Used CMake options:

```
UA_ENABLE_DA=1 UA_ENABLE_DISCOVERY=1 UA_ENABLE_ENCRYPTION=1 UA_ENABLE_ENCRYPTION_OPENSSL=1 
UA_ENABLE_METHODCALLS=1 UA_ENABLE_NODEMANAGEMENT=1 UA_ENABLE_PARSING=1 
UA_ENABLE_SUBSCRIPTIONS=1 UA_LOGLEVEL=100 UA_MULTITHREADING=0
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2
 - [x] Other OPC UA SDKs used (client or server): Prosys OPC UA Simulation Server
 - [x] Operating system: Windows 7
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

## Solution
This appears to be due to the the way that UA_Client_Subscriptions_backgroundPublish works. It creates a bunch of pending publish requests and increments client->currentlyOutStandingPublishRequests until it reaches the configured limit for the connection. This is fine when everything is working as the counter will be decremented when the server sends through a response, but when the connection is interrupted there are a bunch of pending requests left in limbo. And due to this line:

`while (client->currentlyOutStandingPublishRequests < client->config.outStandingPublishRequests)`

it will never send another publish request once the connection is reestablished. Hence the constant 'Inactivity' message being printed out.

I'm pretty sure that client->currentlyOutStandingPublishRequests should be set to 0 whenever a connection is established. That should fix the problem.

StalderT, 2021-08-28:
You can try using the example provided in https://github.com/open62541/open62541/blob/master/examples/client_subscription_loop.c

(Re)creating the subscription when the session is activated should work normally. 

https://github.com/open62541/open62541/blob/530e3c0aaffd07d85f260e1713de39bf2fd22181/examples/client_subscription_loop.c#L73-L102

smitha-cgi, 2021-08-29:
> You can try using the example provided in https://github.com/open62541/open62541/blob/master/examples/client_subscription_loop.c
> 
> (Re)creating the subscription when the session is activated should work normally.
> 

This 'sort of' works, in that the client can start getting data again by resubscribing after the connection has been re-established. The problem is that you lose all of the data that was buffered by the server during the comms outage. Putting in the change that I suggested around clearing the currentlyOutStandingPublishRequests would allow the original subscription to be retained, so the buffered data for that subscription will be sent in by the server once the connection is restored.

o6-nbeier, 2026-10-20:
Thank you for the detailed report.

Stale outstanding Publish requests are now cancelled when the SecureChannel closes (see #5219), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier