open62541/open62541 #4504: Is Multithreading on windows supported ?

vladp, 2021-07-03:
I have added my [comment to an already closed issue](https://github.com/open62541/open62541/issues/3493#issuecomment-832410023), where I am running into a problem building the library with Multithreading and windows (the exact command line and git commit hash are in my comment)

Did not get a follow up, so just wanted to check if this combination (windows and multithreading) is meant to to work

Comment is here https://github.com/open62541/open62541/issues/3493#issuecomment-832410023

Thank you

jpfr, 2021-07-06:
Have a look at the definitions here https://github.com/open62541/open62541/blob/master/arch/win32/ua_architecture.h#L155
and adjust them to your local environment.

We are happy to take patches and suggestions.

Furthermore, I'll mark this as "up for grabs" for a developer with a win environment.

junpingfu, 2021-12-16:
Oh, no. I get same error too. Is the problem solved？

junpingfu, 2021-12-16:
I use the version is 1.3-rc1

vladp, 2021-12-26:
@junpingfu Unfortunately, I did not have time to work on this.
I do not think it is solved yet, although I had not tested the master branch for about 6 months. 
As you can see above, the project accepts contributions in this area.  I think the work is not very hard, as all the architectural layering and segregation of platform code is already done for us.  Perhaps special care needs to be taken in testing.

JGrothoff, 2022-09-28:
Hi, we faced the same problems for v.1.3.3 compiling with Multithreading under VS 2022 in Windows:
* `pthread.h` was included in [ua_log_stdout.c](https://github.com/open62541/open62541/blob/464966135e5cc7af30d9a3f6bdc81be937b95ac3/plugins/ua_log_stdout.c#L63)
* missing definition of `_InterlockedExchangeSub` in [architecture_definitions.h](https://github.com/open62541/open62541/blob/464966135e5cc7af30d9a3f6bdc81be937b95ac3/include/open62541/architecture_definitions.h#L422)
* more missing definitions of `_InterlockedExchange...` if compiled with `UA_FORCE_CPP`

This worked for us:
* cherry-picking commit 3075db12b74873cdd9b846f410001277efa7cb26 on top of v1.3.3 to get rid of pthreads
* replacing `_InterlockedExchangeSub` with `InterlockedExchangeSubtract`
  *  if `InterlockedExchangeSubtract` is still not defined, it can also be replaced by `InterlockedExchangeAdd(addr, - (long) decrease)`
* (replacing `_InterlockedExchange...` with `InterlockedExchange...` as the macros seem to be more appropriate)
* Moving **winsock.h** and **windows.h** includes from [arch/win32/ua_architecture.h](https://github.com/open62541/open62541/blob/cb6fd614daea6a0372e3585b9008d03cb4240fc5/arch/win32/ua_architecture.h#L42-L43) to [include/open62541/architecture_definitions.h](https://github.com/open62541/open62541/blob/cb6fd614daea6a0372e3585b9008d03cb4240fc5/include/open62541/architecture_definitions.h#L403)
  * This was necessary, because architecture_definitions were included before windows.h following the cyclic include path:
    * include/open62541/architecture_definitions.h
    * open62541/config.h
    * win32/ua_architecture.h
  * It's not sufficient to reorder the includes in [arch/win32/ua_architecture.h](https://github.com/open62541/open62541/blob/cb6fd614daea6a0372e3585b9008d03cb4240fc5/arch/win32/ua_architecture.h#L38-L45), because during amalagamation **include/open62541/architecture_definitions.h** will be placed first.
  

We applied these changes in this [fork branch](https://github.com/JGrothoff/open62541/tree/patch_multithreading_win).
This is a hotfix which will hopefully become unnecessary in the future, because of commit 3075db12b74873cdd9b846f410001277efa7cb26 and 108b8c537b2b7edb28038162ffe109c08715dabe from master.

 @jfpr: can you tell whether and when these commits will become part of the next minor or major release?