open62541/open62541 #4509: Tutorial on Events : error during execution

jackybek, 2021-07-04:
## Description

## Background Information / Reproduction Steps
In addNewEventType(UA_-Server *uaServer) tutorial, the following function call return BadNodeIdExist : 

UA_Server_addObjectTypeNode(uaServer, UA_NODEID_NULL, …)

Even when I try to replace the second parameter with ‘UA_NODEID_NUMERIC(0,UA_NS0ID_FOLDERTYPE), …), the error is the same.

Used CMake options:
Events option is turned on

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.2.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: raspian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2021-07-05:
An object type has to be a part of the type hierarchy.
You can't add an object type underneath the folder object.

Please provide a bit more context so we can help.

jackybek, 2021-07-05:
This is where I found the sample tutorial on github:

https://github.com/open62541/open62541/blob/master/examples/tutorial_server_events.c

The error occurred in this function:
static UA_StatusCode
addNewEventType(UA_Server *server) {
    UA_ObjectTypeAttributes attr = UA_ObjectTypeAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "SimpleEventType");
    attr.description = UA_LOCALIZEDTEXT("en-US", "The simple event type we created");
    return UA_Server_addObjectTypeNode(server, UA_NODEID_NULL,
                                       UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE),
                                       UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
                                       UA_QUALIFIEDNAME(0, "SimpleEventType"),
                                       attr, NULL, &eventType);
} 

Does this mean this tutorial has errors?

jackybek, 2021-07-05:
I found the error.. it is due to a repeated nodeId used somewhere in the tree.

jackybek, 2021-07-05:
Now that I have managed to clear all the runtime errors, i am wondering how can I test and validate the Events capability?

jackybek, 2021-07-11:
I tested using UAExpert

closed 2021-07-11 by jackybek