open62541/open62541 #4510: Compile Errors for open62541 on XMC4700

Maku-chan, 2021-07-04:
## Description

## Background Information / Reproduction Steps

I pulled one week ago the open62541 latest version present at the Master branch. I am trying to compile the example tutorial_server_firststeps.c on Dave platform for XMC 4700. I have used the arm-none-eabi-gcc compiler and installed the open62541 package with the below mentioned command.

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
```

But I am getting the below mentioned errors

C:/open62541/build/src_generated/open62541/types_generated_handling.h:8054:35: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 static UA_INLINE UA_WriteResponse *
                                   ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8059:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_StatusCode
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8060:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'UA_WriteResponse_copy'
 UA_WriteResponse_copy(const UA_WriteResponse *src, UA_WriteResponse *dst) {
 ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8064:15: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 UA_DEPRECATED static UA_INLINE void
               ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8069:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8074:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8080:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8085:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_HistoryUpdateDetails *
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8085:42: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 static UA_INLINE UA_HistoryUpdateDetails *
                                          ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8090:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_StatusCode
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8091:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'UA_HistoryUpdateDetails_copy'
 UA_HistoryUpdateDetails_copy(const UA_HistoryUpdateDetails *src, UA_HistoryUpdateDetails *dst) {
 ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8095:15: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 UA_DEPRECATED static UA_INLINE void
               ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8100:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8105:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8111:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8116:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_HistoryUpdateType *
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8116:39: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 static UA_INLINE UA_HistoryUpdateType *
                                       ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8121:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_StatusCode
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8122:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'UA_HistoryUpdateType_copy'
 UA_HistoryUpdateType_copy(const UA_HistoryUpdateType *src, UA_HistoryUpdateType *dst) {
 ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8126:15: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 UA_DEPRECATED static UA_INLINE void
               ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8131:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8136:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8142:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8147:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_PerformUpdateType *
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8147:39: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 static UA_INLINE UA_PerformUpdateType *
                                       ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8152:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_StatusCode
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8153:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'UA_PerformUpdateType_copy'
 UA_PerformUpdateType_copy(const UA_PerformUpdateType *src, UA_PerformUpdateType *dst) {
 ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8157:15: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 UA_DEPRECATED static UA_INLINE void
               ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8162:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8167:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8173:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8178:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_UpdateDataDetails *
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8178:39: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
 static UA_INLINE UA_UpdateDataDetails *
                                       ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8183:8: error: unknown type name 'UA_INLINE'
 static UA_INLINE UA_StatusCode
        ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8184:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'UA_UpdateDataDetails_copy'
 UA_UpdateDataDetails_copy(const UA_UpdateDataDetails *src, UA_UpdateDataDetails *dst) {
 ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8188:15: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'static'
 UA_DEPRECATED static UA_INLINE void
               ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8193:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8198:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA_INLINE void
                  ^
C:/open62541/build/src_generated/open62541/types_generated_handling.h:8204:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 static UA

jpfr, 2021-07-05:
That's the problem: `unknown type name 'UA_INLINE'`
UA_INLINE is defined here: https://github.com/open62541/open62541/blob/master/include/open62541/architecture_definitions.h#L205

Hopefully this helps debugging the problem.

Maku-chan, 2021-07-08:
ok thanks a lot.

closed 2021-08-08 by Maku-chan