open62541/open62541 #4513: Nested custom data types on the client side

nugatritter, 2021-07-06:
## Description
I am coding an OPC UA client module.
Working with simple custom data types works well for me.<br>
But if custom data types have :exclamation:nested:exclamation: custom data types, I have no idea how to deal with that.

## Environment
- open62541 v1.2.2
- Ubuntu 20.04 LTS (64bit) / Windows 10 21H1 (64bit)
- [Automation Demo OPC UA server](https://www.unified-automation.com) 
- wanted OPC UA variable to read from client: ns=2;s=Demo.Static.Scalar.WorkOrder
- the full test case is [here](https://github.com/nugatritter/ExtendedObjectOpen62541/)

## Background Information / Reproduction Steps
I receive the following custom data type information from the server:
**Dictionary information**
```
<opc:TypeDictionary xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:tns="http://www.unifiedautomation.com/DemoServer/" DefaultByteOrder="LittleEndian" xmlns:opc="http://opcfoundation.org/BinarySchema/" xmlns:ua="http://opcfoundation.org/UA/" TargetNamespace="http://www.unifiedautomation.com/DemoServer/">
   <opc:Import Namespace="http://opcfoundation.org/UA/"/>
   ...
   <opc:StructuredType BaseType="ua:ExtensionObject" Name="WorkOrderStatusType">
      <opc:Field TypeName="opc:CharArray" Name="Actor"/>
      <opc:Field TypeName="opc:DateTime" Name="Timestamp"/>
      <opc:Field TypeName="ua:LocalizedText" Name="Comment"/>
   </opc:StructuredType>
   <opc:StructuredType BaseType="ua:ExtensionObject" Name="WorkOrderType">
      <opc:Field TypeName="opc:Guid" Name="ID"/>
      <opc:Field TypeName="opc:CharArray" Name="AssetID"/>
      <opc:Field TypeName="opc:DateTime" Name="StartTime"/>
      <opc:Field TypeName="opc:Int32" Name="NoOfStatusComments"/>
      <opc:Field LengthField="NoOfStatusComments" TypeName="tns:WorkOrderStatusType" Name="StatusComments"/>
   </opc:StructuredType>
   ...
</opc:TypeDictionary>
```

**Browse information**
[image]
[image]

**Example code used (simplified with hard-coded data type for debugging)**
```
typedef struct {
    UA_String actor;
    UA_DateTime timestamp;
    UA_LocalizedText comment;
} sWorkOderStatusType;

typedef struct {
    UA_Guid id;
    UA_String assetId;
    UA_DateTime startTime;
    size_t wostSize;
    sWorkOderStatusType* statusComments;
} sWorkOrderType;

static UA_DataTypeMember WorkOrderType_members[4] = {
    {   // 16 bytes
        UA_TYPES_GUID,            // index
        0,                        // padding
        true,                     // namespace zero
        false,                    // is an array
        false                     // is an optional field
        UA_TYPENAME("ID")         // name
    },
    {   // 16 bytes
        UA_TYPES_STRING,          // index
        offsetof(sWorkOrderType, assetId) - offsetof(sWorkOrderType, id) - sizeof(UA_Guid), // padding = 0
        true,                     // namespace zero
        false,                    // is an array
        false                     // is an optional field
        UA_TYPENAME("AssetID")    // name
    },
    {   // 8 bytes
        UA_TYPES_DATETIME,        // index
        offsetof(sWorkOrderType, startTime) - offsetof(sWorkOrderType, assetId) - sizeof(UA_String), // padding = 0
        true,                     // namespace zero
        false,                    // is an array
        false                     // is an optional field
        UA_TYPENAME("StartTime")  // name
    },
    {   // 16 bytes (array!)
        UA_TYPES_EXTENSIONOBJECT, // index
        offsetof(sWorkOrderType, wostSize) - offsetof(sWorkOrderType, startTime) - sizeof(UA_DateTime), // padding = 0
        true,                     // namespace zero
        true,                     // is an array
        false                     // is an optional field
        UA_TYPENAME("StatusComments")
    }
};

static const UA_DataType WorkOrderType = {
    {2, UA_NODEIDTYPE_NUMERIC, {3003}}, // type ID
    {2, UA_NODEIDTYPE_NUMERIC, {5013}}, // binary encoding ID
    sizeof(sWorkOrderType),             // memSize = 56
    21,                                 // type index
    UA_DATATYPEKIND_STRUCTURE,          // type kind
    false,                              // is pointer free
    false,                              // is overlayable
    4,                                  // members size
    WorkOrderType_members               // member array
    UA_TYPENAME("WorkOrderType")        // name
};

UA_DataTypeArray customDataTypeTestArray {
    0x0,                                // next
    1,                                  // members size
    &WorkOrderType                      // member address
};

UA_Client* client = UA_Client_new();
UA_ClientConfig_setDefault(UA_Client_getConfig(client));
UA_Client_connect(client, uaUrl);
UA_Client_getConfig(client)->customDataTypes = &customDataTypeTestArray;
UA_NodeId nodeId = UA_NODEID_STRING_ALLOC(2, "Demo.Static.Scalar.WorkOrder");
UA_Variant outValue;
UA_StatusCode retval = UA_Client_readValueAttribute(client, nodeId, &outValue);
```
**Returned values**
retval is BadConnectionClosed (0x80AE0000)<br>
the log messages are:
- Error receiving the response with status code BadInternalError
- Connection 332 | SecureChannel 3168226539 | Receiving the response failed with StatusCode BadInternalError
- Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good

## Suspected problem
The fourth member of custom data type WorkOrderType has the index = UA_TYPES_EXTENSIONOBJECT and namespaceZero=true.<br>I guess there should be something else here. Unfortunately,I don't know what. <br>
Do you have any ideas how to solve the problem?

jpfr, 2021-07-06:
We have run into exactly that problem where companion specs were cross-using their datatype definitions.

Update to the latest master branch.
There you don't have to fiddle with indizes anymore.
Instead, you can just set a pointer to the datatype definition for every struct member.

nugatritter, 2021-07-07:
After migrating my module to the latest master branch, everything works perfectly now. Many thanks!

closed 2021-07-07 by nugatritter