open62541/open62541 #4514: Can I reassign the size of an array on the fly?  If so, then when can I do this?

Demolishun, 2021-07-06:
## Description
Can I reassign the size of an array on the fly?  If so, then when can I do this?

I can see that there is a read function, but I would think at that point the datapoint is defined.
I thought I saw reference to some sort of pre read function, but I cannot find documentation on this.

I want to do what is described here, but on the fly.  Initially I would start with an empty array.  (length 0)
https://github.com/open62541/open62541/issues/2455#issuecomment-465613354

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.2 release

jpfr, 2021-07-06:
What do you mean by "in the fly"?
During the initial creation of the node or afterwards?

In general the Write service takes care of all your writing needs...

If the constraints of the node permit it, you can write resize the array any time.

Demolishun, 2021-07-06:
Afterwards.
I have a set of message strings that increase up to a max number.  I want to grow this array and report the number of entries in the array starting with zero entries.  This will change when new messages show up.  So maybe I can do this when that event occurs.  
I think I found the call I was looking for in the backend:
UA_Server_setVariableNode_valueBackend
This allows me to set a callback that is called right before a read occurs.  I found an example in:
server_pubsub_rt_field_information_model.c

Demolishun, 2021-07-07:
Okay, the place to do this is using these callbacks:
https://open62541.org/doc/current/tutorial_server_datasource.html#variable-value-callback

o6-nbeier, 2026-10-20:
Thank you for the question.

The array length is part of the Value, so write a replacement Value with the new length whenever it changes (or update it in the read callback) and keep ArrayDimensions as the upper bound.

Closing for now.

closed 2026-10-20 by o6-nbeier