open62541/open62541 #4515: Memory errors when connect to server witfh Sign&Encrypt and response must be divide to chunks.

nokem61, 2021-07-07:
## Description
When connect to server with Sign&Encrypt and response is bigger then buffersize we have several memory errors witch clear ByteStrings. 

## Background Information / Reproduction Steps
I attached source codes. 
Set buffersize to 10000 and connect via Prosys OPC UA Browser generate problems.
Problem is only when data are encrypted. 
[image]
[image]

Used CMake options:

```bash
cmake -G "Visual Studio 15 2017 Win64" ..\.. -DCMAKE_INSTALL_PREFIX:PATH="../install_64_debug" -DCMAKE_BUILD_TYPE:STRING="Debug" -DUA_ARCHITECTURE:STRING="win32" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL="1" -DUA_ENABLE_ENCRYPTION_OPENSSL:BOOL="1" -DOPENSSL_INCLUDE_DIR:PATH="OpenSSL/include64" -DSSL_EAY_RELEASE:FILEPATH="../../OpenSSL/bin/x64/Release/libssl" -DLIB_EAY_DEBUG:FILEPATH="../../OpenSSL/bin/x64/Debug/libcrypto" -DSSL_EAY_DEBUG:FILEPATH="../../OpenSSL/bin/x64/Debug/libssl" -DLIB_EAY_RELEASE:FILEPATH="../../OpenSSL/bin/x64/Release/libcrypto" -DUA_ENABLE_HISTORIZING:BOOL="1" -DBUILD_SHARED_LIBS:BOOL="1" -DUA_LOGLEVEL:STRING="100" -DUA_ENABLE_DISCOVERY:BOOL="1"  -DUA_DEBUG:BOOL="1" -DUA_ENABLE_DISCOVERY_MULTICAST:BOOL="1" -DCMAKE_SYSTEM_VERSION="10.0.17763.0"
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2.2
 - [x] Other OPC UA SDKs used (client or server): Prosys OPC UA Browser
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached  [OpcuaTests.zip](https://github.com/open62541/open62541/files/6774402/OpcuaTests.zip)
 - [x] Critical issue

nokem61, 2021-07-20:
...

o6-nbeier, 2026-10-20:
Thank you for the report.

The double padding caused by OpenSSL was removed in #5438, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier