open62541/open62541 #4519: build open62541 for 32 bit arm architecture stm32H743 issue

Maku-chan, 2021-07-08:
## Description

## Background Information / Reproduction Steps
I am trying to build the library for 32 bit arm architecture but getting the following error

-- The C compiler identification is GNU 9.3.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/usr/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/maku-chan/opc_ua_application/open62541/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_d86f2/fast && /usr/bin/make -f CMakeFiles/cmTC_d86f2.dir/build.make CMakeFiles/cmTC_d86f2.dir/build
    make[1]: Entering directory '/home/maku-chan/opc_ua_application/open62541/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_d86f2.dir/testCCompiler.c.o
    /usr/bin/arm-none-eabi-gcc    -o CMakeFiles/cmTC_d86f2.dir/testCCompiler.c.o   -c /home/maku-chan/opc_ua_application/open62541/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_d86f2
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d86f2.dir/link.txt --verbose=1
    /usr/bin/arm-none-eabi-gcc      CMakeFiles/cmTC_d86f2.dir/testCCompiler.c.o  -o cmTC_d86f2 
    /home/maku-chan/opc_ua_application/gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /home/maku-chan/opc_ua_application/gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
    exit.c:(.text.exit+0x2c): undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_d86f2.dir/build.make:87: cmTC_d86f2] Error 1
    make[1]: Leaving directory '/home/maku-chan/opc_ua_application/open62541/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_d86f2/fast] Error 2
    

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:5 (project)

Used CMake options:

```bash
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DCMAKE_C_COMPILER=arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=arm-none-eabi-g++ 
```
I think the error is because of the -specs="nosys.specs" file. I want to ask that where can I add this in the cmake file to compile it. Also how can I specify the 32 bit architecture.

Thanks

Maku-chan, 2021-07-08:
I am able to add the flag
´´´bash
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DCMAKE_C_COMPILER=/home/maku-chan/opc_ua_application/arm_gcc/gcc-arm-10.2-2020.11-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc -DCMAKE_C_FLAGS="--specs=nosys.specs"        
´´´
But receive the follow failed commands
-- open62541 Version: v1.2.2-334-gb82ad4df-dirty
-- The selected architecture is: posix
-- Test CC flag -std=c99
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -pipe
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wall
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wextra
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wpedantic
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Werror
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-static-in-inline
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-overlength-strings
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wno-unused-parameter
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wc++-compat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-security
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wformat-nonliteral
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wmissing-prototypes
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wstrict-prototypes
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wredundant-decls
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wuninitialized
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Winit-self
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wcast-qual
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wstrict-overflow
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
-- Test CC flag -Wnested-externs
-- Performing Test flag_supported
-- Performing Test flag_supported - Success
[... 48 lines cut ...]
/home/maku-chan/opc_ua_application/open62541/include/open62541/util.h:194:35: note: format string is defined here
  194 | #define UA_PRINTF_GUID_FORMAT "%08x-%04x-%04x-%02x%02x-%02x%02x%02x%02x%02x%02x"
      |                                ~~~^
      |                                   |
      |                                   unsigned int
      |                                %08lx
/home/maku-chan/opc_ua_application/open62541/src/ua_types_print.c: In function 'printGuid':
/home/maku-chan/opc_ua_application/open62541/include/open62541/util.h:194:31: error: format '%x' expects argument of type 'unsigned int', but argument 4 has type 'UA_UInt32' {aka 'const long unsigned int'} [-Werror=format=]
  194 | #define UA_PRINTF_GUID_FORMAT "%08x-%04x-%04x-%02x%02x-%02x%02x%02x%02x%02x%02x"
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/maku-chan/opc_ua_application/open62541/include/open62541/util.h:194:31: note: in definition of macro 'UA_PRINTF_GUID_FORMAT'
  194 | #define UA_PRINTF_GUID_FORMAT "%08x-%04x-%04x-%02x%02x-%02x%02x%02x%02x%02x%02x"
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/maku-chan/opc_ua_application/open62541/include/open62541/util.h:194:35: note: format string is defined here
  194 | #define UA_PRINTF_GUID_FORMAT "%08x-%04x-%04x-%02x%02x-%02x%02x%02x%02x%02x%02x"
      |                                ~~~^
      |                                   |
      |                                   unsigned int
      |                                %08lx
/home/maku-chan/opc_ua_application/open62541/src/ua_types_print.c: At top level:
cc1: note: unrecognized command-line option '-Wno-static-in-inline' may have been intended to silence earlier diagnostics
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/open62541-object.dir/build.make:147: CMakeFiles/open62541-object.dir/src/ua_types_print.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:383: CMakeFiles/open62541-object.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Maku-chan, 2021-07-08:
I compiled it after changing the gcc compiler to arm-none-linux-gnueabihf-gcc

But now when I run it in the eclipse and link the library I get the below mentioned error

01:32:10 **** Incremental Build of configuration Debug for project test_arm_c_static_project ****
make all 
Building file: ../main.c
Invoking: GNU Arm Cross C Compiler
arm-none-linux-gnueabihf-gcc -mcpu=cortex-m7 -mthumb -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -I/home/maku-chan/opc_ua_application/open62541/include -I/home/maku-chan/opc_ua_application/open62541/build/src_generated -I/home/maku-chan/opc_ua_application/open62541/arch -I/home/maku-chan/opc_ua_application/open62541/deps -I/home/maku-chan/opc_ua_application/open62541/plugins/include -std=c99 -MMD -MP -MF"main.d" -MT"main.o" -c -o "main.o" "../main.c"
Finished building: ../main.c
 
Building target: test_arm_c_static_project.elf
Invoking: GNU Arm Cross C Linker
arm-none-linux-gnueabihf-gcc -mcpu=cortex-m7 -mthumb -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Xlinker --gc-sections -L/home/maku-chan/opc_ua_application/open62541/build/bin -Wl,-Map,"test_arm_c_static_project.map" -o "test_arm_c_static_project.elf"  ./main.o   -lopen62541
during IPA pass: icf
In function ‘UA_LOG_FATAL’:
lto1: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
lto-wrapper: fatal error: arm-none-linux-gnueabihf-gcc returned 1 exit status
compilation terminated.
/home/maku-chan/opc_ua_application/arm_gcc/gcc-arm-10.2-2020.11-x86_64-arm-none-linux-gnueabihf/bin/../lib/gcc/arm-none-linux-gnueabihf/10.2.1/../../../../arm-none-linux-gnueabihf/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [makefile:57: test_arm_c_static_project.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

01:32:11 Build Failed. 2 errors, 0 warnings. (took 671ms)

ChaseWwwwind, 2021-12-14:
Is there any way to compile the file of the arm architecture?

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier