open62541/open62541 #4521: weak UA_NodeId_hash

nugatritter, 2021-07-09:
## UA_NodeId_hash should return a unique number
I use hashes as ID to quickly find complex data in a map (like in typical databases).
But the resulting hash number of UA_NodeId is to weak to do that.

## Environment
- open62541 v1.2.2 #1180574
- Windows 10 21H1 (64bit)

## Reproduction Steps
example code snipped:
```
// map of user defined data types
// the key is the hash number of the node id of the data type
std::map<UA_UInt32, UA_DataType> dataTypeMap;

// data type #1 has id ns=3;i=3009
UA_NodeId id1 = UA_NODEID_NUMERIC(3, 3009);
UA_UInt32 idHash1 = UA_NodeId_hash(&id1);

// data type #2 has id ns=4;i=3014
UA_NodeId id2 = UA_NODEID_NUMERIC(4, 3014);
UA_UInt32 idHash2 = UA_NodeId_hash(&id2);

// idHash1 = 2044113391
// idHash2 = 2044113391
// my dataTypeMap is being corrupted!
```

## Question
Is there any better solution for the key of the dataTypeMap?

nugatritter, 2021-07-10:
How about this solution. This works for me.
open62541/src/ua_types.c
````
// https://www.programmingalgorithms.com/algorithm/sdbm-hash/cpp/
u32
UA_ByteString_hash(u32 fnv, const u8* buf, size_t size) {
    for (size_t i = 0; i < size; ++i) {
        fnv = buf[i] + (fnv << 6) + (fnv << 16) - fnv;
    }
    return fnv;
}

jpfr, 2021-07-11:
A hash-map is not just a map.
If the hash matches, you still have to do a full comparison of the key.

There are many more possible NodeIds than uint32 numbers.
So you cannot avoid the possibility of hash collisions.

Is this anecdotal evidence or can you provide numbers on how common collisions are?
Looking at the code, it looks quite solid.
https://github.com/open62541/open62541/blob/master/src/ua_types.c#L413

The sdbm hash looks good. And doesn't require a multiplication!
So we might switch to that. However, that will not solve the problem of collisions being possible.

nugatritter, 2021-07-12:
At the public OPC UA server "opc.tcp://opcuademo.sterfive.com:26543" I found following when scanning the children of Node ID "/Simulation/Static":
- 87 variables total
- 46 failures when UA_NodeId_hash is used to verify node IDs
- zero failures when using the proposed sdbm hash
Here comes the quick check code (failures only):
```
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3014)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3009)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3014)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3015)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(6, 3006)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3014)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3013)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3010)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3009)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3021)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(5,3023)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3013)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3009)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3009)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(6,3008)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3007)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3012)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3015)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(7, 3019)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3016)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3021)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3022)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3012)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3015)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3017)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3016)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3017)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3018)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3011)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3008)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3023)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3020)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3019)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3016)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3004)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3006)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3005)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3001)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3002)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3010)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(6,3008)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3027)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3024)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3019)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3017)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3007)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3005)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3011)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(6,3008)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3016)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3018)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3015)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3009)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3004)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3006)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3005)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(3,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3020)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3022)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3012)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3013)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3006)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3004)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3021)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3023)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3022)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3020)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3008)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3011)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3020)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3023)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3018)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3017)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3002)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(5,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3026)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3025)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(4, 3007)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3004)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3008)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3009)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3002)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(4,3003)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3013)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3015)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3009)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3011)));
 printf("%d\n",UA_NodeId_hash(&UA_NODEID_NUMERIC(5, 3018)) == UA_NodeId_hash(&UA_NODEID_NUMERIC(7,3016)));
```

nugatritter, 2021-07-12:
Here is the result of a generic test of UA_NodeId_hash. In the table are equal hashes only!
[equal_hash_table_UA_NodeId_hash.pdf](https://github.com/open62541/open62541/files/6801232/equal_hash_table_UA_NodeId_hash.pdf)

nugatritter, 2021-07-12:
I run the same test with modified UA_ByteString_hash (see above) and zero equal hashes were found.

jpfr, 2021-07-12:
Ok, cool. We are switching to sdbm via #4524.

closed 2021-07-12 by nugatritter