open62541/open62541 #4525: CcMake .. error when enabling FULL NameSpaceZero and UA_ENABLE_SUBSCRIPTIONS_EVENTS

jackybek, 2021-07-12:
## Description
How to resolve CMake Error : add_subdirectory given source "doc".
I already run : git submodule update --init --recursive

## Background Information / Reproduction Steps
 CMake Warning at CMakeLists.txt:447 (MESSAGE):
   UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS is enabled.  The feature is under
   development and marked as EXPERIMENTAL

 CMake Error at CMakeLists.txt:1308 (add_subdirectory):
   add_subdirectory given source "doc" which is not an existing directory.

Used CMake options:

BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
BIBTEX_COMPILER:FILEPATH=BIBTEX_COMPILER-NOTFOUND
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CLANG_FORMAT_PROGRAM:FILEPATH=CLANG_FORMAT_PROGRAM-NOTFOUND
CLANG_TIDY_PROGRAM:FILEPATH=CLANG_TIDY_PROGRAM-NOTFOUND
CMAKE_AR:FILEPATH=/usr/bin/ar
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_COLOR_MAKEFILE:BOOL=ON
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_CXX_FLAGS:STRING=
CMAKE_CXX_FLAGS_DEBUG:STRING=-g
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_CXX_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_CXX_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-8
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-8
CMAKE_C_FLAGS:STRING=
CMAKE_C_FLAGS_DEBUG:STRING=-g
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG
CMAKE_EXE_LINKER_FLAGS:STRING=
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_EXE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_EXE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_EXE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_EXPORT_COMPILE_COMMANDS:BOOL=OFF
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
CMAKE_INSTALL_LOCALEDIR:PATH=
CMAKE_INSTALL_LOCALSTATEDIR:PATH=var
CMAKE_INSTALL_MANDIR:PATH=
CMAKE_INSTALL_OLDINCLUDEDIR:PATH=/usr/include
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_INSTALL_RUNSTATEDIR:PATH=
CMAKE_INSTALL_SBINDIR:PATH=sbin
CMAKE_INSTALL_SHAREDSTATEDIR:PATH=com
CMAKE_INSTALL_SYSCONFDIR:PATH=etc
CMAKE_LINKER:FILEPATH=/usr/bin/ld
CMAKE_MAKE_PROGRAM:FILEPATH=/usr/bin/make
CMAKE_MODULE_LINKER_FLAGS:STRING=
CMAKE_MODULE_LINKER_FLAGS_DEBUG:STRING=
CMAKE_MODULE_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELEASE:STRING=
CMAKE_MODULE_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_NM:FILEPATH=/usr/bin/nm
CMAKE_OBJCOPY:FILEPATH=/usr/bin/objcopy
CMAKE_OBJDUMP:FILEPATH=/usr/bin/objdump
CMAKE_RANLIB:FILEPATH=/usr/bin/ranlib
CMAKE_SHARED_LINKER_FLAGS:STRING=
CMAKE_SHARED_LINKER_FLAGS_DEBUG:STRING=
CMAKE_SHARED_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELEASE:STRING=
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_SKIP_INSTALL_RPATH:BOOL=NO
CMAKE_SKIP_RPATH:BOOL=NO
CMAKE_STATIC_LINKER_FLAGS:STRING=
CMAKE_STATIC_LINKER_FLAGS_DEBUG:STRING=
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELEASE:STRING=
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO:STRING=
CMAKE_STRIP:FILEPATH=/usr/bin/strip
CMAKE_VERBOSE_MAKEFILE:BOOL=FALSE
CPACK_SOURCE_RPM:BOOL=OFF
CPACK_SOURCE_TBZ2:BOOL=ON
CPACK_SOURCE_TGZ:BOOL=ON
CPACK_SOURCE_TXZ:BOOL=ON
CPACK_SOURCE_TZ:BOOL=ON
CPACK_SOURCE_ZIP:BOOL=OFF
DVIPDF_CONVERTER:FILEPATH=/usr/bin/dvipdf
DVIPS_CONVERTER:FILEPATH=DVIPS_CONVERTER-NOTFOUND
GIT_EXECUTABLE:FILEPATH=/usr/bin/git
HTLATEX_COMPILER:FILEPATH=HTLATEX_COMPILER-NOTFOUND
LATEX2HTML_CONVERTER:FILEPATH=LATEX2HTML_CONVERTER-NOTFOUND
LATEX_COMPILER:FILEPATH=LATEX_COMPILER-NOTFOUND
LIBWEBSOCKETS_INCLUDE_DIR:PATH=/usr/local/include
LIBWEBSOCKETS_LIBRARIES:FILEPATH=/usr/local/lib/libwebsockets.so
LUALATEX_COMPILER:FILEPATH=LUALATEX_COMPILER-NOTFOUND
MAKEINDEX_COMPILER:FILEPATH=MAKEINDEX_COMPILER-NOTFOUND
MDNSD_LOGLEVEL:STRING=300
OPENSSL_CRYPTO_LIBRARY:FILEPATH=/usr/local/ssl/lib/libcrypto.so
OPENSSL_INCLUDE_DIR:PATH=/usr/local/ssl/include
OPENSSL_SSL_LIBRARY:FILEPATH=/usr/local/ssl/lib/libssl.so
PDFLATEX_COMPILER:FILEPATH=PDFLATEX_COMPILER-NOTFOUND
PDFTOPS_CONVERTER:FILEPATH=/usr/bin/pdftops
PKG_CONFIG_EXECUTABLE:FILEPATH=/usr/bin/pkg-config
PS2PDF_CONVERTER:FILEPATH=/usr/bin/ps2pdf14
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
SPHINX_EXECUTABLE:FILEPATH=/usr/bin/sphinx-build
UA_AMALGAMATION_ARCHITECTURES:STRING=
UA_ARCHITECTURE:STRING=posix
UA_ARCH_ADD_FLAGS:STRING=
UA_ARCH_EXTRA_INCLUDES:STRING=
UA_ARCH_LINKER_FLAGS:STRING=
UA_ARCH_REMOVE_FLAGS:STRING=
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_FUZZING:BOOL=OFF
UA_BUILD_FUZZING_CORPUS:BOOL=OFF
UA_BUILD_OSS_FUZZ:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_DEBUG:BOOL=OFF
UA_DEBUG_DUMP_PKGS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_CLANG_COV:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_ENCRYPTION_MBEDTLS:BOOL=OFF
UA_ENABLE_ENCRYPTION_OPENSSL:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=OFF
UA_ENABLE_PARSING:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP_XDP:BOOL=OFF
UA_ENABLE_PUBSUB_FILE_CONFIG:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_PUBSUB_MONITORING:BOOL=ON
UA_ENABLE_PUBSUB_MQTT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_TYPEDESCRIPTION:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=ON
UA_FILE_NS0:BOOL=OFF
UA_FILE_NS0_BLACKLIST:BOOL=OFF
UA_FORCE_32BIT:BOOL=OFF
UA_FORCE_CPP:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_PACK_DEBIAN:BOOL=OFF
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): v1.2-rc1-20-g78a6721b
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ X] Operating system: Raspbian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ X] Critical issue

jackybek, 2021-07-13:
Latest results after running "make"

pi@OPCServer-109:~/open62541/build $ make
[  4%] Built target libmdnsd
[ 12%] Built target open62541-generator-statuscode
[ 16%] Built target open62541-generator-transport
[ 20%] Built target open62541-generator-types
[ 24%] Built target open62541-generator-namespace
[ 40%] Built target open62541-amalgamation-header
[ 56%] Built target open62541-amalgamation-source
[ 56%] Built target open62541-code-generation
[ 88%] Built target open62541-object
[ 92%] Built target open62541
[ 96%] Building C object tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o
**/home/pi/open62541/tools/ua2json/ua2json.c:21:10: fatal error: open62541/types.h: No such file or directory
 #include <open62541/types.h>**
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [tools/ua2json/CMakeFiles/ua2json.dir/build.make:63: tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:788: tools/ua2json/CMakeFiles/ua2json.dir/all] Error 2

I found the file "types.h> in github and downloaded into my environment.  I then try to make again.. this time round it complain:
[ 96%] Building C object tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o
**In file included from /home/pi/open62541/tools/ua2json/ua2json.c:21:
/usr/local/include/open62541/types.h:19:10: fatal error: open62541/config.h: No such file or directory
 #include <open62541/config.h>**
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [tools/ua2json/CMakeFiles/ua2json.dir/build.make:63: tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:788: tools/ua2json/CMakeFiles/ua2json.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

The questions are :
1. Where can I find the file "config.h"
2. In "types.h", it make references to 2 other header files : "common.h" and "statuscodes.h".  Where can I find in github?

jpfr, 2021-07-14:
See in the folder <build>/src_generated/ generated at build-time.
Do you have an old version of open62541 installed system-wide?

The compiler can get confused if there is an old version in /usr/include or similar location with higher precedence.

jackybek, 2021-07-14:
I have installed different versions of open62541 files in the same rpi under different directories.  Based on what you mentioned about the compiler getting confused, it seemed that I can only install 1 version.

jackybek, 2021-07-15:
Ok .. I did something simpler : turn off cmake option:
1. UA_ENABLE_DISCOVERY_MULTICAST
2. UA_ENABLE_DISCOVERY_SEMAPHOR

the previous errors are gone now.
However the compiler now says : open62541.c: UA_DataType has no member named typeIndex;

these errors appeared in the following functions:
1. UA_PubSubChannelMQTT_regist
2. UA_PubSubChannelMQTT_unregist
3. UA_PubSubChannelMQTT_send

jackybek, 2021-07-15:
There is also Another error: struct lws_context_creation_info has no member named ‘ws_ping_pong_interval’ in function ServerNetworkLayerWS_start

So for the moment I just comment out this line.
Please advise how to go about solving the typeIndex error mentioned above.  Thanks.

jackybek, 2021-07-15:
After i made a minor correction due to a space between # & pragma, the above errors disappeared.
I am now left with the following include errors.
Where can I find this header file ? ua_architecture.h>
------------------------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------------------------
[ 91%] Built target open62541
[ 95%] Building C object tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o
In file included from /usr/local/include/open62541/types.h:19,
                 from /home/pi/open62541/tools/ua2json/ua2json.c:21:
/usr/local/include/open62541/config.h:118:10: fatal error: posix/ua_architecture.h: No such file or directory
 #include "posix/ua_architecture.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [tools/ua2json/CMakeFiles/ua2json.dir/build.make:63: tools/ua2json/CMakeFiles/ua2json.dir/ua2json.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:751: tools/ua2json/CMakeFiles/ua2json.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

jackybek, 2021-07-17:
I did a fresh installation using : git clone https://github.com/open62541/open62541.git
cmake ..
ccmake ..
make

Then build my source code.  All the above errors are resolved.

However, i get a segmentation fault when adding a new variable node to the OPCUA tree.  Previously i did not choose FULL namespace zero and it runs perfect.  Now with FULL namespace zero it crashed.
---------------------------------------------------------------------
        UA_UInt16 namespaceIndex=1;         // 0 is reserved for OPC

            // Add objectnode to represent Airgard->Timestamp
        UA_NodeId r2_airgard_timestamp_Id;  /* get the nodeid assigned by the server */
        UA_ObjectAttributes oAttr_r2_timestamp = UA_ObjectAttributes_default;
        UA_Server_addObjectNode(uaServer, UA_NODEID_STRING(namespaceIndex, "AirGardSensor_TimeStamp"),
            r1_airgardroot_Id,                  // Make Airgard node <Contains> Timestamp node          //UA_NODEID_NUM$
            UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
            UA_QUALIFIEDNAME(namespaceIndex, "Timestamp"),      //UA_QUALIFIEDNAME(1, "Timestamp"),
            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
            oAttr_r2_timestamp, NULL, &r2_airgard_timestamp_Id);
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
            "Object Node created : Airgard->Timestamp Node");   //, r1_airgardroot_Id);

        // variable
        UA_VariableAttributes vInstrumentTimeAttr = UA_VariableAttributes_default;
        vInstrumentTimeAttr.description = UA_LOCALIZEDTEXT("en-US", "Instrument Time Info");
        vInstrumentTimeAttr.displayName = UA_LOCALIZEDTEXT("en-US", "1. Instrument Time");
        vInstrumentTimeAttr.dataType = UA_TYPES[UA_TYPES_STRING].typeId;
        vInstrumentTimeAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
        vInstrumentTimeAttr.historizing = false;
        //UA_String InstrumentTime = (UA_String) XMLInstrumentTime;
        UA_String InstrumentTime[MAX_STRING_SIZE];
        //strcpy(InstrumenTime, XMLInstrumentTIme);
        //UA_Variant_setArray(&vInstrumentTimeAttr.value, &InstrumentTime, MAX_STRING_SIZE, &UA_TYPES[UA_TYPES_STRING]);
        UA_Variant_setScalar(&vInstrumentTimeAttr.value, &InstrumentTime, &UA_TYPES[UA_TYPES_STRING]);
        UA_Server_addVariableNode(uaServer, UA_NODEID_NUMERIC(namespaceIndex, 10101),   **// fail here !!!**
            r2_airgard_timestamp_Id,
            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
            UA_QUALIFIEDNAME(namespaceIndex, "Instrument Time"),        //UA_QUALIFIEDNAME(1, "Instrument Time"),
            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
            vInstrumentTimeAttr, NULL, &outInstrumentTime_Id);
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
            "Attribute Variable created: Airgard->TimestampType->Instrument Time");     //, r2_airgard_timestamp_Id);

----------------------------------
For full name space zero setup, what other things do I need to perform?

jackybek, 2021-07-17:
All errors are resolved.
Seemed that 2 nodes in the OPC tree is corrupted during creation process.
After i comment out these 2 nodes, my program runs with FULL_NAMESPACE.

closed 2021-07-17 by jackybek