open62541/open62541 #4528: Handling multiple definitions of the same Method

mlgiraud, 2021-07-13:
## Description

When trying to add a method callback to a method, one has to add the method callback to each instance of the method.
Consider for example the AddCertificate method `(NodeId="i=12548" BrowseName="AddCertificate")` which is part of the TrustListType `NodeId="i=12522"`.
The ua-nodeset specifies an instance for example in the TrustList (`NodeId="i=12642"`) which is part of the DefaultApplicationGroup (`NodeId="i=14156"`) in the ServerConfiguration object.
From a logical point i would expect the methods in both TrustLists to reference the same method callback, since the context is determined by the object the method is called on (like a `this` pointer in c++).
However, when registering the method callback on the type definition, the callback will not exist on the instantiated TrustList, since the namespace generator code creates a new method node for the instantiated TrustList.

The nodeset files contain a MethodDeclarationId that references the original method node from the type definition, however the standard only mentions this in https://reference.opcfoundation.org/Core/docs/Part6/F.9/ which says
> May be specified for Method Nodes that are a target of a HasComponent reference from a single Object Node. It is the NodeId of the UAMethod with the same BrowseName contained in the TypeDefinition associated with the Object Node.
If the TypeDefinition overrides a Method inherited from a base ObjectType then this attribute shall reference the Method Node in the subtype.

Example from the nodeset file:
```
<UAMethod NodeId="i=12668" BrowseName="AddCertificate" ParentNodeId="i=12642" MethodDeclarationId="i=12548">
```

Would it be sensible to not create these new method nodes and reference the original method?
If not, maybe we could link the callbacks, such that only the method callback of the original type definition has to be set.
A possible solution could also be to include an internal reference to the method declaration id and look for a callback there if no callback was set on the new method node, making it possible to actually override the method in the type definition (and even call the original callback in the overridden callback maybe).

jpfr, 2021-07-14:
The nodeset compiler should be able to do both.
Generate a dedicated method or set a reference to an existing method node.

Just to be sure to understand which it is:
- Your nodeset xml defines several method nodes (with pre-defined NodeIds) as **part of object instances**, but you want to not generate some of them and reuse others instead.
- Your nodeset xml defines an **object type with methods** and you don't want to make a full copy of the methods in the type when generating instances.

mlgiraud, 2021-07-14:
The point is that the nodeset file specifies the method nodes explicitly as new nodes, bute references the original method node with the `MethodDeclarationId`, which is currently ignored. It would be useful to be able to register the callback once on the method referenced by `MethodDeclarationId`, and then be able to use it in the newly created method instances, if their callback is not explicitly overridden. Because otherwise i would have to register the same callback for every node defined this way in the nodeset (I really don't know why the opc foundation? chose to model it this way, instead of relying on the automatic creation by type instantiation).