open62541/open62541 #4541: Removal of Dictionaries-Node from Namespace Zero

iosb-ina-mr, 2021-07-19:
## Description

The [I4AAS companion specification](https://opcfoundation.org/markets-collaboration/i4aas/) depends on the Dictionaries-Node from namespace zero. Due to that node being removed in [src/server/ua_server_ns0.c](https://github.com/open62541/open62541/blob/e1f8fd3b18dd237c1660df79eae411c250348250/src/server/ua_server_ns0.c#L1014), the companion specification does not load properly.
I wonder why the Dictionaries-Node is removed in the first place along with several other nodes? According to the commit history, those nodes were removed for a micro-embedded-device profile. This profile was removed in February but the nodes, however, are still removed.

## Background Information
[Commit: Removal of nodes for micro-embedded profile](https://github.com/open62541/open62541/commit/508aacb9589a82e4bcdf71e45a2a7673c42c1328#diff-78df7143ea0c91caf7c6e0969cd003ad1265cbdb359a2fa529228c5263734dfe)
[Commit: Removal of micro-embedded profile](https://github.com/open62541/open62541/commit/bdb233e810a7c9f5187c2e53f021c71abdefba80#diff-78df7143ea0c91caf7c6e0969cd003ad1265cbdb359a2fa529228c5263734dfe)

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL ..
```

## Checklist

 - open62541 Version (master-branch: e1f8fd3b18dd237c1660df79eae411c250348250):
 - Operating system: Ubuntu 20.04 (WSL)

jpfr, 2021-07-21:
Sure, we can leave the dictionary node in for the reduced profile.
Would you provide a PR?

iosb-ina-mr, 2021-07-21:
Sure, I have provided a PR for this issue.

NoelGraf, 2021-07-23:
#4550 is merged.

closed 2021-07-23 by NoelGraf